Sloeber/arduino-eclipse-plugin #578: Sloeber 3.1 does not show splash screen

nicoverduin, 2016-11-04:
I had to reinstall Windows 7 last week and rebuild my development environment. My Data disk is a separate one. I installed sloeber 3.1 (again) in the root of my D drive and to my surprise, no more splash screen. The rest worked without problems.
I had 3.0 in another folder and that works fine including the splash screen. 
It is not a serious problem, but I don't like issues without a cause explanation. And of course not  a priority :).
So a clean Windows 7 installation (including all updates)->Sloeber 3.1 in Data drive (D:\)->No splash screen.

And btw in this folder : D:\sloeber\configuration\org.eclipse.equinox.launcher\it.baeyens.arduino.application_3.1.0.201608031424

the is a splash screen present. Just not showing.

Regards
Nico

jantje, 2016-11-05:
Nico thanks for reporting.
this is a duplicate of #488.

closed 2016-11-05 by jantje

nicoverduin, 2016-11-05:
Hoi Jantje

Klopt inderdaad. Ff een oude splash (3.0) erin gezet en werkt als een tierelier. .

Daar ik in de laatste overnight build elke keer die toolchain moet veranderen wacht ik ff af als je het niet erg vind. (wel waarschijnlijk :))

Grtz

Nico