Sloeber/arduino-eclipse-plugin #579: How to call esp8266 SDK APIs?

comebackgb, 2016-11-09:
I tried to call esp8266 SDK APIs，but could not  combiner.It's works fine on Arduino IDE.How to fixed?

`#include "Arduino.h"

//#include "Streaming.h"

extern "C"
{
#include "ets_sys.h"
#include "os_type.h"
#include "osapi.h"
#include "mem.h"
#include "user_interface.h"
#include "cont.h"
}

// Counter
uint32_t i = 0;
// Temporary buffer
uint32_t b = 0;

int lap = 0;

//The setup function is called once at startup of the sketch
void setup()
{
// Add your initialization code here
  Serial.begin(115200);
  // Call API-Function
  system_deep_sleep_set_option(4);
  // Call API-Function
  system_rtc_mem_read(64,&b,4);
  // Check if the value is within expected range
  if(b < 100000)
    i = b;
}

// The loop function is called in an endless loop
void loop()
{
  for(int c = 0; c < 50; c++)
  {
    //Serial << i++ << " " << lap << endl;
    delay(10);
  }
  // Call API-Function
  system_rtc_mem_write(64,&i,4);
  delay(1000);
  // Call API-Function
  system_deep_sleep(10000000);
  lap++;
}
`
Errors:
Starting combiner
"/Applications/Eclipse.app/Contents/Eclipse/arduinoPlugin/tools/esp8266/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-L/Applications/Eclipse.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lib" "-L/Applications/Eclipse.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/ld" "-Teagle.flash.4m1m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy -o "/Users/ComeBack/Documents/workspace_c/test8266_new/Release/test8266_new.elf" -Wl,--start-group   ./.ino.cpp.o   arduino.ar  "/Users/ComeBack/Documents/workspace_c/test8266_new/Release/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group "-L/Users/ComeBack/Documents/workspace_c/test8266_new/Release"
./.ino.cpp.o:(.text.setup+0x14): undefined reference to `system_deep_sleep_set_option(unsigned char)'
./.ino.cpp.o:(.text.setup+0x18): undefined reference to `system_rtc_mem_read(unsigned char, void*, unsigned short)'
./.ino.cpp.o: In function `HardwareSerial::begin(unsigned long)':
/Applications/Eclipse.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/HardwareSerial.h:75: undefined reference to `system_deep_sleep_set_option(unsigned char)'
./.ino.cpp.o: In function `setup':
/Users/XXX/Documents/workspace/test8266/Release/../test8266.ino:30: undefined reference to `system_rtc_mem_read(unsigned char, void*, unsigned short)'
./.ino.cpp.o:(.text.loop+0x8): undefined reference to `system_rtc_mem_write(unsigned char, void const*, unsigned short)'
./.ino.cpp.o:(.text.loop+0xc): undefined reference to `system_deep_sleep(unsigned int)'
./.ino.cpp.o: In function `loop':
/Users/XXX/Documents/workspace/test8266/Release/../test8266.ino:42: undefined reference to `system_rtc_mem_write(unsigned char, void const*, unsigned short)'
/Users/XXX/Documents/workspace/test8266/Release/../test8266_new.ino:45: undefined reference to `system_deep_sleep(unsigned int)'
collect2: error: ld returned 1 exit status
make: *** [test8266_new.elf] Error 1

freeck, 2016-11-09:
I had the same issue, was reported a while ago on esp8266.com.
Work around is to edit ino.cpp and you will see that the extern "C" part is not there.  Then copy the extern-stuff to ino.cpp, and then build again. It worked for me.

Freeck

comebackgb, 2016-11-09:
@freeck it's works,thanks.

jantje, 2016-11-09:
I would advice not to use .ino files if you need extern c.
The .ino.cpp file is regennerated each time you save the .ino file. As such it will undo the changes you made in the .ino.cpp file.
I see 2 possible solutions to this issue:
1) use .cpp file instead of .ino file.
2) create a new header file that contains the extern C includes and include this one in your .ino file.

freeck, 2016-11-09:
Ok, thanks!

closed 2017-01-17 by jantje