Sloeber/arduino-eclipse-plugin #585: add support for burning bootloader

jantje, 2016-11-21:
The bootloader burning is the only core functionality Arduino IDE has but Sloeber does not yet have.

kgyrtkirk, 2017-08-06:
yes :) it would be awesome :D
anyway...sloeber is cool as is already...this feature is nowhere a showstopper :)
I'll just workaround this somehow :)

wimjongman, 2020-02-09:
This is needed to change the clock on the ATtiny, my new best friend. Any idea what we need to do here?

jantje, 2020-02-09:
@wimjongman 
I'm not sure what you are trying to do as there are already attiny packages.
But to add the bootloader is mainly finding the right environment variables to run the right comands in a console.
From what I see you need A.BOOTLOADER.TOOL to get the tool and
and A.TOOLS.[TOOLNAME].BOOTLOADER.PATTERN to get the command
I guess there is more to it but this is a good starting point

wimjongman, 2020-02-10:
The clock speed of the ATtiny is factory set to 1MHZ. It can only be changed to 8/16mhz with burn bootloader.

I have used the Arduino IDE to do that and that does the trick but obviously I want Sloeber to do that.

You say it is a good starting point but I have no idea what you are talking about.

jantje, 2020-02-10:
If you look at project properties->C/C++ variables->environment you will find a lot of environment variables starting with A. and some with JANTJE. and some others not from Sloeber
The A. are "arduino provided variables" these are from the boards.txt and platform.txt capitalized and A. put in front.
There are also some defaults from https://github.com/Sloeber/arduino-eclipse-plugin/tree/master/io.sloeber.core/config
The JANTJE. stuff is created by Sloeber mostly used to redirect.

In the list you should find A.BOOTLOADER.TOOL which below is avrdude
[image]

If I look for  A.TOOLS.AVRDUDE.BOOTLOADER.PATTERN I get
[image]
Which is the command to run on the command line

It would be great if you could use this environment variable in a launch configuration but I havn't figured out a way to do so.

Code wise knowing the project and the environment name Common.getBuildEnvironmentVariable can return you the value

wimjongman, 2020-02-10:
Thanks, that makes a lot of sense.

jantje, 2021-07-27:
I opened a issue on CDT to make these environment variables available in the "external tools" https://bugs.eclipse.org/bugs/show_bug.cgi?id=575037
I already implemented the functionality and I ran into a issue https://github.com/arduino/arduino-cli/issues/1368

It  seems the bootloader is a 2 step process https://arduino.github.io/arduino-cli/latest/platform-specification/#burn-bootloader
I'm not sure how to deal with that in "external tools"
I guess I should have a look at how to do this in make

jantje, 2021-08-11:
The CDT thing is not going to fly so I implemented it in Sloeber #1367 
I figured out the Arduino issue. Seems the programmer.txt is needed to burn a bootloader

I'm currently considering adding make targets to the makefile. This will not cater for all boards but it probably will for some. It does solve the 2 step process and will be easier than external tools to configure.

jantje, 2021-08-23:
I'm implementing a disconnect from serial port if the build target is listed in a gui manageable list and if the project contains a com port and if the com port is connected in the serial monitor.

closed 2021-09-10 by jantje