Sloeber/arduino-eclipse-plugin #586: cant build project for MightyCore in sloeber

yoavshtainer, 2016-11-22:
first of all, thank you for your great work!

i try sloeber few month ago and cant make it work for me, and today i got it work.
i try with uno and esp8266 and it's all work good, but when i try to use MightyCore i get error in building the project (hello world).

i'm a rookie in eclipse so i don't understand the hole thing.

my error is:

`13:58:54 **** Clean-only build of configuration Release for project dsfsdf ****
"C:\sloeber8\sloeber\arduinoPlugin\tools\make\make" clean
make: *** No rule to make target 'clean'. Stop.

13:58:55 Build Finished (took 758ms)

13:58:56 **** Build of configuration Release for project dsfsdf ****
"C:\sloeber8\sloeber\arduinoPlugin\tools\make\make" all
'Building file: ../.ino.cpp'
'Starting C++ compile'
"/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284p -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -I"C:\sloeber8\sloeber\arduinoPlugin\packages\MightyCore\hardware\avr\1.0.7\cores\MightyCore" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp" -o ".ino.cpp.o" -Wall
subdir.mk:24: recipe for target '.ino.cpp.o' failed
process_begin: CreateProcess(NULL, sh.exe -c "/bin/avr-g++ -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284p -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -IC:\sloeber8\sloeber\arduinoPlugin\packages\MightyCore\hardware\avr\1.0.7\cores\MightyCore -MMD -MP -MF.ino.cpp.d -MT.ino.cpp.o -D__IN_ECLIPSE__=1 -x c++ ../.ino.cpp -o .ino.cpp.o -Wall", ...) failed.
make (e=2): The system cannot find the file specified.

make: *** [.ino.cpp.o] Error 2

13:58:57 Build Finished (took 1s.110ms)`

i don't know how to look about the board.txt file or understand what going on in there, so if you can help me i will be very happy.

thank a lot!

jantje, 2016-11-22:
There have been some fixes in this area. Are you on the nightly?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

yoavshtainer, 2016-11-22:
thank you for your quick answer!

i'm using stable V3.1_win64.2016-08-03_16-31-05.tar

jantje, 2016-11-22:
Won't work with stable

closed 2016-11-24 by jantje