Sloeber/arduino-eclipse-plugin #589: ESP32 on sloeber?

thl-mot, 2016-12-02:
Hi,
I just downloaded sloeber V 3.1 and the nightly build. Is there a way to use the ESP32 with the IDE? In Arduino-IDE there's the arduino-esp32 (https://github.com/espressif/arduino-esp32), is there a way to integrate that manually to sloeber-ide?
Thomas

jantje, 2016-12-02:
The only way to find out is try. Your best shot is to install and test it to the arduino IDE and then add the arduino/hardware/espressief folder to the private hardware path.
Note that lot of issues have been fixed in regards to "other boards" in V4 (still not released as stable)

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

thl-mot, 2016-12-03:
Hi Jantje,
Is there a specification, how "other boards" are stored in your ide. Does it differ between V3.1 and V4.x?
In 3.1 I found two directories that have parts of the information of the board:
1. arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/...
2. arduinoPlugin/tools/...
In 4.x there is no tools directory under arduinoPlugin
Thomas

jantje, 2016-12-03:
Point 2 is as to #558
You should not change thing in [eclipse Install]/arduinoPlugin
Use  "private hardware path" to point to any (and as many as you want) location on your local disks

thl-mot, 2016-12-05:
ok, now it works fine

closed 2016-12-05 by thl-mot

abkglobal, 2017-04-25:
Please can you explain the steps of  how you added ESP32 to sloeber and got it working?