Sloeber/arduino-eclipse-plugin #593: V4 esp8266 gdbstub did not work , now it does...

freeck, 2016-12-07:
Gdbstub combined with "Platform and Boards" (esp8266) version 2.0 and higher did not work.
It seems that the latest uart-functions compared to the older version (esp8266->1.6.5-947-g39819f0) behave differently.
So the  workaround is to use boards.txt version "1.6.5-947-g39819f0".

jantje, 2016-12-07:
Thanks for the input :-)

freeck, 2017-01-05:
I reported this issue at https://github.com/esp8266/Arduino/issues/2663
But unfortunately no response yet. Any suggestions how to tackle this issue would be welcome....

jantje, 2017-01-05:
If I understand this is a issue for my hardware repository to state that only 1.6.5 is supported

freeck, 2017-01-05:
Yes, correct.

When I get some inspiration I will digg into this problem once again, although this issue seems to be not an high priority one as nobody responses...

jantje, 2017-01-05:
That is one of the disadvantages of being a early bird ;-)

freeck, 2017-06-05:
Finally got some inspiration. Version 2.3 is working now (for 90%).
I overlooked a statement in my version of file gdbstub-cfg.h 
#ifndef GDBSTUB_CTRLC_BREAK
#define GDBSTUB_CTRLC_BREAK 0   // this was set to 1 in version of release V1.6 and worked fine so far.
#endif_
Is  working for 90% means that there is still a (minor) issue: the debug console output is not redirected to the Eclipse console. It seems that function "os_install_putc1(gdb_semihost_putchar1)" is not working or is overridden by somewhere else.

jantje, 2017-06-05:
Sounds good
You mean changing this line?

jantje, 2017-06-05:
Some stuff has changed in cdt as well as to default enabled or not in regards to shown debug info.
So that may be a root cause as well

freeck, 2017-06-05:
Sure, set the define to 0.

jantje, 2017-06-05:
Will do when I get home

freeck, 2017-10-10:
[https://github.com/Sloeber/arduino-eclipse-plugin/issues/830#issuecomment-335477997](url)

> However the ets_printf messages are sent to the serial port , so with a second serial port of my PC I am able to observe the messages.

jantje, 2017-10-12:
I still havn't changed to 0 :-(

jantje, 2017-10-12:
Seems I can't get it to work anymore.  I get this all the time
```
Error in final launch sequence
Failed to execute MI command:
-target-select remote \\.\\COM29
Error message from debugger back end:
Bogus trace status reply from target: timeout
Bogus trace status reply from target: timeout
```
With 1 or 0

jantje, 2017-10-12:
Did a commit for this one

closed 2017-10-12 by jantje

freeck, 2017-10-12:
When you get a message in de debugconsole like "...bogus information...." or something like that; check wether the gdbstub is realy installed in your target. You can check this by viewing the output of the serial port of the target: it should, after a reset, send debugmessage $T05#B9
And dont forget to set the buildoptions to -ggdb -Og

jantje, 2017-10-12:
@freeck 
Thanks but ... I first need to find my nodemcu as running nodemcu code on a lonin board might be the issue as well :-(