Sloeber/arduino-eclipse-plugin #595: Indexer doesn't get predefined macros.

jief666, 2016-12-08:
The command to get compiler specs is wrong, causing tons of displayed errors which aren't. The command shouldn't have "-o" inside. Cdt expects macros to be outputed on stdout.
I replaces the command by the same as $COMMAND without '-o" : "${A.COMPILER.PATH}${A.COMPILER.CPP.CMD}" ${A.COMPILER.CPP.FLAGS} -mmcu=${A.BUILD.MCU} -DF_CPU=${A.BUILD.F_CPU} -DARDUINO=${A.RUNTIME.IDE.VERSION} -DARDUINO_${A.BUILD.BOARD} -DARDUINO_ARCH_${A.BUILD.ARCH} ${A.COMPILER.CPP.EXTRA_FLAGS} ${A.BUILD.EXTRA_FLAGS} ${A.INCLUDES} -E -P -v -dD "${INPUTS}"
and it's fine.

jantje, 2016-12-08:
Are you running V4 with the latest update?

jantje, 2016-12-08:
I looked here on windows and the -o is there and all works fine.
Which OS are you on?

jantje, 2016-12-08:
oh I see it is only the macro's that are not working.

jantje, 2016-12-08:
Should be fixed with update of tomorrow

jief666, 2016-12-08:
I've just install this today. I'm on Mac OS X Yosemite. I installed through eclipse update site. How will I get the latest update ?

jantje, 2016-12-08:
What version does the splash screen say

jief666, 2016-12-08:
Eclipse Neon-1

jantje, 2016-12-08:
Naturally, you used the update site.
Can you check the version of the sloeber plugin?
I really would like to know as version 3.1 should not have the -o and V4 you should not have via the update site

jief666, 2016-12-09:
The V3 has the bug which prevent reset on Leonardo.
So I went for V4 nightly build when I found that a correction was committed (on November I think). Easier for me than rebuild everything from source !
The version I have in my eclipse Neon-1 is  4.0.0.201612080213.
The URL is http://eclipse.baeyens.it/update/V4/nightly . Maybe it's not what is called an update site. I went to "Install new software" in Eclipse and put that url.

jief666, 2016-12-09:
I've just tried "Check for update in Eclipse" and Eclipse found today's version. So it works. Thanks.

jantje, 2016-12-09:
Which means you are on V4 :-)
Note that __IN_Eclipse__ seems still missing.

jantje, 2017-01-01:
closing as fixed and it was regression. The IN_ECLIPSE is fixed as weel

closed 2017-01-01 by jantje