Sloeber/arduino-eclipse-plugin #599: Error with eclipse but not with arduino IDE

luca120, 2016-12-11:
Hi all, I write here because i can't fix my issue on eclipse with arduino plugin, I'm using the ubuntu, when I  do starting the code with Arduino IDE I haven't any issue, but if I begin compiling the same code on ecplise I see the error after have pushed the verify button

below I have copied all error code, can you help me for fix this issue?

Thank you in advance

`00:22:03 **** Incremental Build of configuration Release for project prova ****
make all 
Building file: ../.ino.cpp
Starting C++ compile
"/home/luca/Scaricati/eclipse/arduinoPlugin/tools/arduino/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR     -I"/home/luca/Scaricati/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.15/cores/arduino" -I"/home/luca/Scaricati/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.15/variants/eightanaloginputs" -I"/home/luca/Scaricati/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.15/libraries/SPI" -I"/home/luca/Scaricati/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.15/libraries/SPI/src" -I"/home/luca/Documenti/Librerie/SD" -I"/home/luca/Documenti/Librerie/SD/utility" -I"/home/luca/Documenti/Librerie/NewliquidCrystal" -I"/home/luca/Documenti/Librerie/NewliquidCrystal/utility" -I"/home/luca/Documenti/Librerie/DallasTemperature" -I"/home/luca/Documenti/Librerie/PID" -I"/home/luca/Documenti/Librerie/Buttons" -I"/home/luca/Documenti/Librerie/MAX6675" -I"/home/luca/Scaricati/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.15/libraries/Wire" -I"/home/luca/Scaricati/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.15/libraries/Wire/src" -I"/home/luca/Documenti/Librerie/OneWire" -I"/home/luca/Documenti/Librerie/Keypad" -I"/home/luca/Documenti/Librerie/Keypad/utility" -I"/home/luca/Scaricati/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.15/libraries/EEPROM" -I"/home/luca/Scaricati/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.15/libraries/EEPROM/src" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
In file included from ../prova.ino:12:0,
                 from ../.ino.cpp:21:
/home/luca/Documenti/Librerie/MAX6675/max6675.h:10:7: error: redefinition of 'class MAX6675'
 class MAX6675 {
       ^
In file included from ../.ino.cpp:14:0:
/home/luca/Documenti/Librerie/MAX6675/max6675.h:10:7: error: previous definition of 'class MAX6675'
 class MAX6675 {
       ^
In file included from ../.ino.cpp:16:0:
/home/luca/Scaricati/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.15/libraries/EEPROM/src/EEPROM.h:145:20: warning: 'EEPROM' defined but not used [-Wunused-variable]
 static EEPROMClass EEPROM;
                    ^
subdir.mk:24: set di istruzioni per l'obiettivo ".ino.cpp.o" non riuscito
make: *** [.ino.cpp.o] Errore 1`

jantje, 2016-12-11:
I assume this file does not have include safeguards. 
 /home/luca/Documenti/Librerie/MAX6675/max6675.h
this is a known limitation with an easy workaround (which is actually a standard practise in C/C++ development) See issue #271 

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2017-01-01:
close as open for long time without response

closed 2017-01-01 by jantje