Sloeber/arduino-eclipse-plugin #602: Library errors in Eclipse; clean compile with ArduinoIDE

jrobert-gh, 2016-12-16:
I've made the empty .h files in my libraries where Eclipse wants to find them, but I'm getting errors in "included" files I did not include nor need:

"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/tools/arduino/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.12/cores/arduino" -I"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.12/variants/standard" -I"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.12/libraries/SoftwareSerial" -I"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.12/libraries/SoftwareSerial/src" -I"/Users/jrobert/Projects/ArduinoWksp/libraries/Blynk" -I"/Users/jrobert/Projects/ArduinoWksp/libraries/Blynk/src" -I"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/SD/1.0.9" -I"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/SD/1.0.9/src" -I"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/Ethernet/1.1.2" -I"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/Ethernet/1.1.2/src" -I"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/WiFi/1.2.6" -I"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/WiFi/1.2.6/src" -I"/Users/jrobert/Projects/ArduinoWksp/libraries/BlynkESP8266_Lib" -MMD -MP -MF"libraries/WiFi/src/utility/spi_drv.cpp.d" -MT"libraries/WiFi/src/utility/spi_drv.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/WiFi/1.2.6/src/utility/spi_drv.cpp" -o "libraries/WiFi/src/utility/spi_drv.cpp.o"  -Wall
/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/WiFi/1.2.6/src/utility/spi_drv.cpp:21:17: fatal error: SPI.h: No such file or directory
 #include <SPI.h>

I don't use use the SPI, Ethernet, WifI, or SD libraries in my code, but Eclipse includes them anyway in the gcc command line. My same code compiles with the Arduino IDE. Why are these libraries being included? If they would compile and get optimized out, that would be OK, but they are causing compile errors. SPI.h, for instance, isn't even in my library folder - there is no reason to put it there.

jantje, 2016-12-17:
You can look in the include browser why the are included.

closed 2016-12-17 by jantje

jrobert-gh, 2016-12-17:
I assume "include browser" means "preferences | Arduino | Library
Manager"? If so, I've unchecked everything from there that I don't need.
In fact, there is nothing left checked there at all.

But now the -I list mentions Blynklinux files (which fail the compile
with "No such file or directory") and I don't see anywhere to deselect
them. How did those even get there?

jantje, 2016-12-17:
Include browser is a CDT view that tells you how a file is included. You have the show file includers and the show files included. I made a screengrab where I ask the includers of AES.cpp in a project of mine and you can clearly see why it is part of the project (included in maairobot.h)
[image]

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jrobert-gh, 2016-12-17:
Thanks Jantje, and I am already a patron. Please look me up on Patrion.

jantje, 2016-12-17:
Thanks for being a patron.
I can't make the link between the 2. So it is better to tell me on patreon :-) Than I try to be nicer :-)