Sloeber/arduino-eclipse-plugin #603: add gnu ARM open ocd debug launch

jantje, 2016-12-17:
To have everything installed in Sloeber to do debugging of the zero following 2 features need to be added:
```
GNU ARM C/C++ Cross Compiler

Version: 2.2.1.201606210758
Build id: @build@

Copyright (c) Liviu Ionescu 2009-2014.  All rights reserved.

This plug-in is part of the GNU ARM Eclipse plug-ins project.
For more details, visit <http://gnuarmeclipse.github.io>.
For updates, use <http://gnuarmeclipse.sourceforge.net/updates>.
```

```
GNU ARM C/C++ OpenOCD Debugging

Version: 4.1.1.201606210758
Build id: @build@

Copyright (c) Liviu Ionescu 2009-2014.  All rights reserved.

This plug-in is part of the GNU ARM Eclipse plug-ins project.
For more details, visit <http://gnuarmeclipse.github.io>.
For updates, use <http://gnuarmeclipse.sourceforge.net/updates>.
```

wimjongman, 2016-12-19:
The site [1] does not seem to be valid. I cannot retrieve the repo from that location.

[1] http://gnuarmeclipse.sourceforge.net/updates

Can anyone try to access it from withing Eclipse by adding it as an update site?

[image]

eclipse.buildId=4.7.0.I20161208-0830
java.version=1.8.0_91
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.committers.product -clearpersistedstate
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.committers.product -clean -clearpersistedstate

org.eclipse.equinox.p2.transport.ecf
Warning
Mon Dec 19 14:28:16 CET 2016
Connection to http://gnuarmeclipse.sourceforge.net/updates/p2.index failed on sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target. Retry attempt 0 started

javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.ssl.Alerts.getSSLException(Unknown Source)
	at sun.security.ssl.SSLSocketImpl.fatal(Unknown Source)
	at sun.security.ssl.Handshaker.fatalSE(Unknown Source)
	at sun.security.ssl.Handshaker.fatalSE(Unknown Source)
	at sun.security.ssl.ClientHandshaker.serverCertificate(Unknown Source)
	at sun.security.ssl.ClientHandshaker.processMessage(Unknown Source)
	at sun.security.ssl.Handshaker.processLoop(Unknown Source)
	at sun.security.ssl.Handshaker.process_record(Unknown Source)
	at sun.security.ssl.SSLSocketImpl.readRecord(Unknown Source)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(Unknown Source)
	at sun.security.ssl.SSLSocketImpl.writeRecord(Unknown Source)
	at sun.security.ssl.AppOutputStream.write(Unknown Source)
	at org.apache.http.impl.io.AbstractSessionOutputBuffer.flushBuffer(AbstractSessionOutputBuffer.java:159)
	at org.apache.http.impl.io.AbstractSessionOutputBuffer.flush(AbstractSessionOutputBuffer.java:166)
	at org.apache.http.impl.AbstractHttpClientConnection.doFlush(AbstractHttpClientConnection.java:272)
	at org.apache.http.impl.AbstractHttpClientConnection.flush(AbstractHttpClientConnection.java:277)
	at org.apache.http.impl.conn.AbstractClientConnAdapter.flush(AbstractClientConnAdapter.java:201)
	at org.apache.http.protocol.HttpRequestExecutor.doSendRequest(HttpRequestExecutor.java:239)
	at org.apache.http.protocol.HttpRequestExecutor.execute(HttpRequestExecutor.java:121)
	at org.apache.http.impl.client.DefaultRequestDirector.tryExecute(DefaultRequestDirector.java:685)
	at org.apache.http.impl.client.DefaultRequestDirector.execute(DefaultRequestDirector.java:487)
	at org.apache.http.impl.client.AbstractHttpClient.doExecute(AbstractHttpClient.java:863)
	at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:82)
	at org.eclipse.ecf.provider.filetransfer.httpclient4.HttpClientRetrieveFileTransfer.performConnect(HttpClientRetrieveFileTransfer.java:1084)
	at org.eclipse.ecf.provider.filetransfer.httpclient4.HttpClientRetrieveFileTransfer.access$0(HttpClientRetrieveFileTransfer.java:1075)
	at org.eclipse.ecf.provider.filetransfer.httpclient4.HttpClientRetrieveFileTransfer$1.performFileTransfer(HttpClientRetrieveFileTransfer.java:1071)
	at org.eclipse.ecf.filetransfer.FileTransferJob.run(FileTransferJob.java:74)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:56)
Caused by: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.validator.PKIXValidator.doBuild(Unknown Source)
	at sun.security.validator.PKIXValidator.engineValidate(Unknown Source)
	at sun.security.validator.Validator.validate(Unknown Source)
	at sun.security.ssl.X509TrustManagerImpl.validate(Unknown Source)
	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(Unknown Source)
	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(Unknown Source)
	... 24 more
Caused by: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.provider.certpath.SunCertPathBuilder.build(Unknown Source)
	at sun.security.provider.certpath.SunCertPathBuilder.engineBuild(Unknown Source)
	at java.security.cert.CertPathBuilder.build(Unknown Source)
	... 30 more

jantje, 2016-12-19:
works with me.
Maybe it was a glitch?
[image]

jantje, 2016-12-20:
de nightly build failed with the sourceforge update site as error
Just relaunched it.

jantje, 2016-12-20:
Keeps on  failing
IMHO using sourceforge as download site is just not working properly
I see similar problems with json files with board definitions
```
[INFO] Fetching p2.index from http://download.eclipse.org/releases/neon/201609281000/ (0B at 0B/s)

[INFO] Fetching p2.index from http://download.eclipse.org/releases/neon/201610111000/ (0B at 0B/s)

[INFO] Adding repository http://download.eclipse.org/nebula/snapshot

[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe0cc0a456dcc5d3bd8; expires="Fri, 18-Dec-2026 14:27:44 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:44 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe0cc0a456dcc5d3bd8; expires="Fri, 18-Dec-2026 14:27:45 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:45 GMT"

[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe0cc0a456dcc5d3bd8; expires="Fri, 18-Dec-2026 14:27:45 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:45 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: sf_mirror_attempt="gnuarmeclipse:freefr:Eclipse/updates/p2.index"; expires=120; Path=/". Unable to parse expires attribute: 120

[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe3bf503f395370e4c9; expires="Fri, 18-Dec-2026 14:27:47 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:47 GMT"

[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe3bf503f395370e4c9; expires="Fri, 18-Dec-2026 14:27:47 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:47 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe3bf503f395370e4c9; expires="Fri, 18-Dec-2026 14:27:48 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:48 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: sf_mirror_attempt="gnuarmeclipse:heanet:Eclipse/updates/p2.index"; expires=120; Path=/". Unable to parse expires attribute: 120
[INFO] Adding repository http://gnuarmeclipse.sourceforge.net/updates

[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe6cc0a456dce307f3b; expires="Fri, 18-Dec-2026 14:27:50 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:50 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe6cc0a456dce307f3b; expires="Fri, 18-Dec-2026 14:27:50 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:50 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: sf_mirror_attempt="gnuarmeclipse:netix:Eclipse/updates/content.jar"; expires=120; Path=/". Unable to parse expires attribute: 120

[WARNING] Invalid cookie header: "Set-Cookie: webflash=; expires="Thu, 15-Dec-2016 14:27:52 GMT"; Max-Age=0; Path=/". Unable to parse expires attribute: "Thu, 15-Dec-2016 14:27:52 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: webflash=; expires="Thu, 15-Dec-2016 14:27:52 GMT"; Max-Age=0; Path=/". Unable to parse expires attribute: "Thu, 15-Dec-2016 14:27:52 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: webflash=; expires="Thu, 15-Dec-2016 14:27:52 GMT"; Max-Age=0; Path=/". Unable to parse expires attribute: "Thu, 15-Dec-2016 14:27:52 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: webflash=; expires="Thu, 15-Dec-2016 14:27:52 GMT"; Max-Age=0; Path=/". Unable to parse expires attribute: "Thu, 15-Dec-2016 14:27:52 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe80b4738274d05cc54; expires="Fri, 18-Dec-2026 14:27:52 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:52 GMT"

[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe9ebbb923273706d8b; expires="Fri, 18-Dec-2026 14:27:53 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:53 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe9ebbb923273706d8b; expires="Fri, 18-Dec-2026 14:27:54 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:54 GMT"

[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe9ebbb923273706d8b; expires="Fri, 18-Dec-2026 14:27:54 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:54 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: sf_mirror_attempt="gnuarmeclipse:netix:Eclipse/updates/content.xml.xz"; expires=120; Path=/". Unable to parse expires attribute: 120

[WARNING] Invalid cookie header: "Set-Cookie: webflash=; expires="Thu, 15-Dec-2016 14:27:56 GMT"; Max-Age=0; Path=/". Unable to parse expires attribute: "Thu, 15-Dec-2016 14:27:56 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: webflash=; expires="Thu, 15-Dec-2016 14:27:56 GMT"; Max-Age=0; Path=/". Unable to parse expires attribute: "Thu, 15-Dec-2016 14:27:56 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: webflash=; expires="Thu, 15-Dec-2016 14:27:56 GMT"; Max-Age=0; Path=/". Unable to parse expires attribute: "Thu, 15-Dec-2016 14:27:56 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: webflash=; expires="Thu, 15-Dec-2016 14:27:56 GMT"; Max-Age=0; Path=/". Unable to parse expires attribute: "Thu, 15-Dec-2016 14:27:56 GMT"
[... cut ...]
	at hudson.maven.MavenModuleSetBuild$MavenModuleSetBuildExecution.cleanUp(MavenModuleSetBuild.java:1065)
	at hudson.model.Run.execute(Run.java:1785)
	at hudson.maven.MavenModuleSetBuild.run(MavenModuleSetBuild.java:531)
	at hudson.model.ResourceController.execute(ResourceController.java:98)
	at hudson.model.Executor.run(Executor.java:410)
Caused by: com.sun.mail.smtp.SMTPAddressFailedException: 554 5.7.1 Recipient address rejected: Username jan@baeyens.it and sender nobody@nowhere doesnt match

	at com.sun.mail.smtp.SMTPTransport.rcptTo(SMTPTransport.java:1733)
	... 10 more
Finished: FAILURE

```

llemtt, 2017-01-01:
http://gnuarmeclipse.sourceforge.net/updates doesn't work for me either, I usually download the repository zip and install locally, it works!

jantje, 2017-01-01:
Now there is a copy on sloeber and sloeber uses that copy it should be all fine

ArmandoMobileVoid, 2017-01-06:
Getting the exact same error trying to install GNU ARM Eclipse plugin, tried yesterday with the same problem.

I installed some other plugin to day from source forge and it worked ...so believe its is related to this plugin.

[GNU ARM Wont install.pdf](https://github.com/jantje/arduino-eclipse-plugin/files/690575/GNU.ARM.Wont.install.pdf)

jantje, 2017-01-06:
Apparently you need the very latest jdk or work around by using this update site http://sloeber.io/gnuarmeclipse/stable

ArmandoMobileVoid, 2017-01-09:
Believe the issue was something in Eclipse V 4.4 just would not find it in sourceforge, I installed Mars and it installed fine...(but it did leave off a few things unchecked....). I had the latest idk....installed it before I tried to get this other stuff working. Thanks for the help!

closed 2017-01-17 by jantje