Sloeber/arduino-eclipse-plugin #613: Make default workspace appear in Documents or My Documents

mjmeijer, 2016-12-28:
The default workspace when Sloeber is first starting now appears in the application root folder that is not accessible for beginner type users on Mac. In my view workspace is a document and should appear in the platform default location:

- windows in "\<username\>\My Documents"
- linux in "\<username\>\Documents"
- macosx in "\<username\>\Documents"

This used to be through defining custom config.ini but Maven/Tycho build does not support that.
The proper way is via p2.inf for which I will provide a patch once assigned to me

mjmeijer, 2016-12-29:
Create a p2.inf file that takes care of setting this up during Tycho build and export. Would look like this on Linux and similar on other platforms.
[image]

jantje, 2016-12-30:
ok. Seems like a good idea.

universpharmacie, 2017-01-05:
The advantage of having it in the Documents folder is (if you use macOS Sierra) that everything is automatically backed up / synchronized towards iCloud. Thus: Great idea!

dgolda, 2017-01-07:
Beware of filenames with spaces on Windows! I've got a problem with workspce created in My Documents.
BTW, on Windows 10 there is no more My Documents folder but just Documents.

mjmeijer, 2017-01-07:
Great Tip, we'll change it right @jantje ?

jantje, 2017-01-07:
>Beware of filenames with spaces on Windows!

Actually on all OS'es. Correctly quoting is not so easy and os and tool dependent.

>Great Tip, we'll change it right @jantje ?
How do you propose to change it?

mjmeijer, 2017-01-07:
I opened https://bugs.eclipse.org/bugs/show_bug.cgi?id=510071

mjmeijer, 2017-01-07:
Before release and resolution we can move to @user.home fro win32 only in p2.inf file.

jantje, 2017-01-08:
I propose for now to keep windows in the old way.
I want setup some test environments so I can test spaces.

jantje, 2017-01-23:
I think this is fixed in V4 or am I wrong?

jantje, 2017-01-28:
This is fixed in windows.
@mjmeijer 
Does it work as planned on Mac

closed 2017-05-25 by jantje