Sloeber/arduino-eclipse-plugin #614: esp8266 CallSDKFunctions sample fails in combiner stage when compiled in sloeber

mogoreanu, 2016-12-29:
Repro:
Linux mint 17.2 x64
1. Download fresh sloeber x64 
2. Add esp8266 board 2.3.0
3. Create "New Arduino sketch" 
4. Choose "Sample sketch" > esp8266 > CallSDKFunctions
5. Press Verify

The CallSDKFunctions sketch compiles and works in Arduino IDE, but fails to compile in sloeber with the following error message:
tarting combiner
"/home/mogo/eclipse/sloeber/arduinoPlugin/tools/esp8266/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-L/home/mogo/eclipse/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lib" "-L/home/mogo/eclipse/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/ld" "-Teagle.flash.4m1m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy -o "/home/mogo/eclipse/sloeber/workspaces/esp8266/call_sdk_functions/Release/call_sdk_functions.elf" -Wl,--start-group   ./.ino.cpp.o   arduino.ar  "/home/mogo/eclipse/sloeber/workspaces/esp8266/call_sdk_functions/Release/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group "-L/home/mogo/eclipse/sloeber/workspaces/esp8266/call_sdk_functions/Release"
./.ino.cpp.o: In function `setup':
/home/mogo/eclipse/sloeber/workspaces/esp8266/call_sdk_functions/Release/../CallSDKFunctions.ino:23: undefined reference to `wifi_station_get_hostname()'
./.ino.cpp.o: In function `loop':
/home/mogo/eclipse/sloeber/workspaces/esp8266/call_sdk_functions/Release/../CallSDKFunctions.ino:25: undefined reference to `wifi_station_get_hostname()'
collect2: error: ld returned 1 exit status
make: *** [call_sdk_functions.elf] Error 1

It seems that the esp8266 SDK header files are included correctly, but the library is not linked in.

The contents of the 'lib' directory are:
mogo@andromeda ~/eclipse/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lib $ ls
libairkiss.a  libhal.a       libmesh.a      libsmartconfig.a  libwpa.a
libat.a       libjson.a      libnet80211.a  libssl.a          libwps.a
libaxtls.a    liblwip.a      libphy.a       libstdc++.a       README.md
libcrypto.a   liblwip_gcc.a  libpp.a        libupgrade.a
libespnow.a   libmain.a      libpwm.a       libwpa2.a

P.S. The idea behind sloeber is awesome and I'm supporting you on patreon.

mogoreanu, 2016-12-29:
Attaching sloeber and arduino ide build logs
[arduino_ide_call_std_log.txt](https://github.com/jantje/arduino-eclipse-plugin/files/676657/arduino_ide_call_std_log.txt)
[sloeber_call_std_log.txt](https://github.com/jantje/arduino-eclipse-plugin/files/676658/sloeber_call_std_log.txt)

mogoreanu, 2016-12-29:
Interestingly enough, if I move the code from the sample CallSDKFunctions.ino sketch to a .cpp file in another project it builds and works as expected so I am unblocked.

Nevertheless it would be cool if sample sketches worked out of the box just to avoid confusion. I'd be happy to help if you point me in the direction where the problem might be hiding.

jantje, 2016-12-30:
First of all thanks for your support.
I looked at the code and the problem is:
```
extern "C" {
#include "user_interface.h"
}
```
this is not supported by sloeber.
If you can change user_interface.h I would put the extern"C" in this file. See this link for more details https://isocpp.org/wiki/faq/mixing-c-and-cpp#include-c-hdrs-personal

if you can not change user_interface.h you can create a new header file include that header file and move the include
```
extern "C" {
#include "user_interface.h"
}
```
in your new header file.

duplicate of https://github.com/jantje/arduino-eclipse-plugin/issues/579

mogoreanu, 2016-12-30:
Can we issue an automatic Warning/Error for .ino files that have extern "C" with a link  to the bug? It would have saved me (and I suppose others) a couple hours of debugging.

jantje, 2016-12-30:
I'll see what I can do

jantje, 2017-01-01:
@mjmeijer 
That is the document I used to start to write the code.
Unfortunately the IInclude does not specify wether it is conditionally or not.
Maybe we can find out that extern "C" is used.
If we also look for the first define and the last include line and the first define line is before the last include line we can also report a warning/error

jantje, 2017-01-02:
I found a solution for this problem.
I reopened the original issue so I close this one

closed 2017-01-02 by jantje