Sloeber/arduino-eclipse-plugin #615: New Sketch wizard does not work on Mac

mjmeijer, 2016-12-29:
Both built and from Eclipse the New Sketch Wizard does not work on Mac. Both boards and upload protocol are not filled in. And the Port menu is not resistant to plugging in and out boards while dialog is open.
Als the size is far too wide, where does this come from?
Using null parameter as switch in BoardDescriptor means get from preferences, that are never initialized?

mjmeijer, 2016-12-29:
Does not work from nightly on Linux either...

jantje, 2016-12-30:
I guess something is not behaving as you expect as universpharmacie reports using V4 and he works on mac.
There are size issues in V4. I did some work on hiding none used fields but that seems to cause issues in to many occasions. Doing a resize fixes it.
If only a swt expert could have a look ;-)

universpharmacie, 2016-12-30:
Can not reproduce the issue, at least not for the Teensy processors - I don't work with any others - 8bit is evil! ;-) . After going once into Preferences -> Arduino -> Private hardware path and setting it to '/Applications/Arduino.app/Contents/Java/hardware/teensy/avr', the 'New sketch' wizard works always fine for me (see attached pic)
[image]

jantje, 2016-12-30:
glad to see the gaps between the options are gone on mac as well :-)

jantje, 2017-01-01:
@mjmeijer 
have you been able to find out what causes the confusion/problem?

mjmeijer, 2017-01-01:
I can now get it to work after adding some custom hardware and libraries paths, so it has something to do with PreferenceInitializer I think. Will ook into it later. My patch works on my branch, now get from upstream, rebase and push branch then do pull request right?

jantje, 2017-01-01:
>My patch works on my branch, now get from upstream, rebase and push branch then do pull request right?

Something like that yes ;-)

wimjongman, 2017-01-01:
Yes.⁣

mjmeijer, 2017-01-02:
Out of the box, there is a single boards.txt file and the board file menu is hidden, but as selection is not set the other menus are never initialized. Fix is to force selection after the listeners are in place.
I would like to take the liberty to also do some refactoring and renaming to become more Java-like. OK?

jantje, 2017-01-02:
Be careful when refactoring. This is inherited code that is used in both the wizard and the project properties.
Which means that it fits in 2 "gui integration frameworks" which has it's trick.

As to the problem.
It is quite possible that when only 1 boards.txt file is found the combo is not properly initialized.
This may be so because comparing file names in eclipse is tricky due to the many compatibility options. I had some issues there with the last used stuff so it may have hit us in the "brand new installation" as well.
As we are at it it might be better not to hide the combo. I'm saying this based on feedback I got.

mjmeijer, 2017-01-02:
When running from Eclipse IDE that combo also becomes very wide because of long path length. I have that fixed as well, looking like this:
[image]
I will test it with both new wizard and properties on project, on Mac and Linux before committing.
If you want I can stop hiding the Combo as well.

jantje, 2017-01-02:
>When running from Eclipse IDE that combo also becomes very wide because of long path length. I have that fixed as well, looking like this:

Great. I have been looking at this :-)

>If you want I can stop hiding the Combo as well.

please do

mjmeijer, 2017-01-02:
[image]
Shortening also works on Linux...

closed 2017-01-18 by jantje