Sloeber/arduino-eclipse-plugin #616: avrdude from tools/arduino/avrdude/6.3.0-arduino8/bin broken

gagara, 2016-12-29:
'avrdude' binary from eclipse plugin has broken dependency on ncurses libtinfo:
```
$ ldd avrdude 
	linux-vdso.so.1 (0x00007fffb6fc4000)
	librt.so.1 => /lib64/librt.so.1 (0x00007f90f8751000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f90f8535000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f90f8239000)
	libtinfo.so.5 => not found
	libc.so.6 => /lib64/libc.so.6 (0x00007f90f7ea0000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f90f8959000)
```
Plugin version: 3.0.0.201603312133

jantje, 2016-12-30:
I'm missing what you are trying to say.

gagara, 2016-12-30:
I want to say that avrdude library which is packaged into plugin has external dependencies on system libraries, especially on ncurses-5.x. If you don't have it (in my case I have ncurses-6.x installed) then avrdude just not work and you are not able to upload sketch.
The same true for all other libs which ldd shows as dependencies. If some of them are missing in a system then plugin became unusable.

BTW: is there any easy way to configure plugin to use external gcc and avrdude but not that which are packaged in plugin? In older version of plugin it was possible if I'm not mistaken.

jantje, 2016-12-30:
What wazs confusing me:

>I want to say that avrdude library which is packaged into plugin 

Sloeber does not package a avrdude. The avrdude found at "tools/arduino/avrdude/6.3.0-arduino8/bin" is packaged by arduino.
To be more precise in toolchain 1.6.16 (for avr)  and 1.6.10 (for samd) 
As to you in llinux (windows works fine for a change ;-) ) this version of avrdude is dependent on a OS installed library. If so the latest arduino IDE should come with this library.
Can you check whether the latest Arduino IDE comes with a good library?

jantje, 2016-12-30:
As to this thread it seems as a real Arduino issue https://bbs.archlinux.org/viewtopic.php?id=221230

gagara, 2016-12-30:
yep. Just checked with latest Arduino-1.8.0. The same issue.

jantje, 2016-12-30:
Thanks for checking.
Did you make an issue in the Arduino github?

gagara, 2016-12-30:
yes. just reported here https://github.com/arduino/Arduino/issues/5783

jantje, 2016-12-31:
txs :-)

jantje, 2017-01-05:
The issue moved here https://github.com/arduino/avrdude-build-script/issues/4

jantje, 2017-01-05:
Closing as Arduino needs to fix and this version will never be fixed

closed 2017-01-05 by jantje