Sloeber/arduino-eclipse-plugin #618: SD Library - Initialising SD Card failed - Arduino Due

420579436, 2017-01-01:
I get no compile warnings or errors.
The code runs in the Arduino IDE.
I have replaced the Sloeber SD libraries by the Arduino libraries, but I could not solve the issue.
I have included the pins_arduino.h library too, but the same result.

Has anyone an idea?

Best
Thorsten

jantje, 2017-01-01:
I have no idea.

420579436, 2017-01-02:
The issue is solved, but I'm not really happy.

After hours of searching through the relevant libraries I've replaced all files in the directory sloeber > arduinoPlugin > packages > arduino > hardware > sam by the original Arduino files.

Thorsten

closed 2017-01-02 by 420579436

jantje, 2017-01-02:
>After hours of searching through the relevant libraries I've replaced all files in the directory sloeber > arduinoPlugin > packages > arduino > hardware > sam by the original Arduino files.

If it were not the original files that were there someone/something must have changed them on your system. Sloeber downloads and unpacks the files in exactly the same way Arduino IDE does.

420579436, 2017-01-02:
> If it were not the original files that were there someone/something must have changed them on your system. Sloeber downloads and unpacks the files in exactly the same way Arduino IDE does.

The Sloeber libraries are partially different.
E.g. the SD library: 
Line 71 of the Sloeber sd.h file `boolean begin(uint32_t clock, uint8_t csPin);` 
doesn't exist in my Arduino sd.h file.

The SD library in particular is very badly maintained.

Thorsten

jantje, 2017-01-02:
are you sure that you have set the same version for sd library in arduino ide and sloeber?
We both use the same json files to find the same archives and unpack in the same way.
But you have to specify the version in both so the version may be different.
And that would explain why there is a difference.
Especially as the Arduino IDE warns about new versions and Sloeber does not. So it is likely you have a older version in sloeber or in Arduino IDE because of installed longer ago.

420579436, 2017-01-02:
You are right. I've not used the latest SD library version. I used 1.0.8.
Now I have used the Sloeber SD library and it works.
The reason for my issue lay in one of the 2,500 other files which I have replaced with my Arduino files.
 
Thorsten

jantje, 2017-01-02:
I strongly advice to not change things in the arduinoPlugin Folder. It is not needed and it makes support very hard for me.

Also please support me on patreon https://www.patreon.com/jantje