Sloeber/arduino-eclipse-plugin #625: Sloeber.app is damaged and can't be opened

elledgeg, 2017-01-07:
Hello, I have tried six times over three days using both Firefox and Safari to download the MacOS Latest stable version of Sloeber. Whereas the download succeeds and the tar file expands the resultant app gives the message "Sloeber.app is damaged and can't be opened". I am using MacOS 10.12.2.
Can you help please.
Thanks,
George

jantje, 2017-01-07:
Can you provide the exact link you tried to download?

mjmeijer, 2017-01-07:
Problem is Security settings, open System Preferences Security, allow applications from anywhere and move Sloeber to the Applications folder. Then try again

mjmeijer, 2017-01-07:
Whe should add a readme.txt to the archive

jantje, 2017-01-07:
I think it would be better to update http://eclipse.baeyens.it/installAdvice.shtml
And add a strong link to that page here http://eclipse.baeyens.it/stable.php?OS=MacOS and http://eclipse.baeyens.it/nightly.php
@mjmeijer Do you feel like you have the skills to handle this?
The files to change are 
/website/WebContent/installAdvice.shtml
/website/WebContent/stable.php
/website/WebContent/nightly.php

elledgeg, 2017-01-08:
Hello,

Thanks for your speedy response.

Security settings, as mentioned by Maarten Meijer proved to be the problem. Having done as he suggested, everything went smoothly and I now have Sloeber up and running. One point, MacOS Sierra (10.12.2) no longer offers the “Allow Applications from Anywhere” option, and it is first necessary to enable it via the command line using "sudo spctl --master-disable"

Thanks everyone for your help.

George.

jantje, 2017-01-28:
I did the update to the web pages. Waiting for nightly to publish

closed 2017-01-28 by jantje

elledgeg, 2017-01-29:
Thanks a lot.

jimconrad, 2017-09-06:
I had to use "sudo spctl --master-disable" to get OS X 10.12.6 to offer the "Allow Applications from Anywhere" option, and after successfully starting sloeber the first time, I could return my security settings to "App store and identified developers".  

I used to be an Apple Developer and could be again (I let my membership drop).  Would the sloeber community have any interest in fixing the problem by signing with a Developer ID certificate?

wimjongman, 2017-09-06:
If it can be a developer called Sloeber or an organization called Sloeber then yes.