Sloeber/arduino-eclipse-plugin #627: Serial monitor often doesn't work (3.1)

judge2005, 2017-01-13:
Hi,

I seem to have a lot of problems using the serial monitor. It very really seems to connect to the board, I have tried resetting it, unplugging the board and plugging it back in etc. Sometimes it works, sometimes not.

It would also be useful to be able to just close the connection from Eclipse.

The Arduino IDE serial monitor works fine.

jantje, 2017-01-13:
You can close the connection with the X next to the +
If you do not close the connection and disconnect the board you will have connection problems.
Depending on the OS it goes from annoying to bad

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

judge2005, 2017-01-13:
Ah. The cross is always gray, so I assumed it was not enabled. Clicking it works. Thanks.

jantje, 2017-01-13:
It used yo go red.

jantje, 2017-05-25:
@rlogiacco 
I see this is still not fixed. The icon now goes from light gray to darker gray.
Can you take a look

closed 2017-07-29 by jantje