Sloeber/arduino-eclipse-plugin #629: compiler flags of an assembly file not passed through to the build process

freeck, 2017-01-16:
Using Compile options: -ggdb -Og -mlongcalls  one should see the flags passed through to the build process, normally this happens indeed. However since "the last few weeks" that is not the case anymore ONLY in case I select Platform Folder 1.6.5 (which I need for using the debugger). 
In case I select Platform Folder 2.3 the compiler flag "-mlongcalls" is present!

Output build process:
bsp 1.6: .....-c -g -x assembler-with-cpp -DF_CPU=80000000L ....
bsp 2.3: ..... -c -g -x assembler-with-cpp -mlongcalls -DF_CPU=80000000L .....

Clean installation of V4 on Windows10 did not help....

jantje, 2017-01-16:
I'm not sure what you are trying to say here.
for #610 I changed that the compile options for c and c++ are no longer passed on to the assembly instructions.
As Sloeber is now the only way I know to modify the assembly command is modifying the boardst.txt.
Adding a "add to assembly" compile option would not be that hard.

freeck, 2017-01-16:
Well, that is exactly what is happening since a few weeks, and what I did not expect :).
I will check whether I am able to modify the boards.txt file.....

freeck, 2017-01-16:
You have to modify the platform.txt file. After comparing the platform-files of both bsp's it was clear what was missing in the 1.6.5 platform.txt. I added flag -mlongcalls to the compileroptions.
Thanks for the hint...

jantje, 2017-01-16:
from 1.6.5 platform.txt file
`compiler.S.flags=-c -g -x assembler-with-cpp -MMD`
from 2.3 platform.txt files
`compiler.S.flags=-c -g -x assembler-with-cpp -MMD -mlongcalls`

freeck, 2017-01-16:
Exactly.
Case closed :)

closed 2017-01-16 by freeck