Sloeber/arduino-eclipse-plugin #637: debugger console gdb trace output

freeck, 2017-01-19:
Using the debugger I always was able to watch the debugger console trace window , which next to all gdb messages also displays the ets_printf() messages (add Serial.setDebugOutput(true); in Setup()) .  Release 4.0.0.201612191522 worked fine, however after installing the latest version the debugger console trace window  is no longer visible....

jantje, 2017-01-19:
Hi Freeck
Sloeber does not add anything to the CDT debugger. As such this is a pure CDT issue.
I know CDT did some changes in the consoles as you can see from this issue in the gnuarm eclipse plugins https://github.com/gnuarmeclipse/plug-ins/issues/156
I do not understand what your issue is but as this is all CDT it is very likely you need to look at a different place.
If you think this really is a issue please open a ticket with cdt https://wiki.eclipse.org/CDT/Bugs and tell me so I can discuss in the CDT developers list.

freeck, 2017-01-19:
> I do not understand what your issue is:

See this clip at timestamp 5 minutes and further: 
https://m.youtube.com/watch?feature=youtu.be&v=2A8l7DejygE
As you can see there a "console display selection" option, which shows the output of ets_printf statements.

OK , I understand its not a Sloeber-issue so I will direct this issue to https://wiki.eclipse.org/CDT/Bugs.
I will keep you informed.

freeck, 2017-01-19:
Jan, I see there is already a reference on https://wiki.eclipse.org/CDT/Bugs to this issue. Do I have to take further action? Perhaps add some version information?

freeck, 2017-01-19:
Jan, I see you already made a reference to this issue on https://github.com/gnuarmeclipse/plug-ins/issues/156
Do I have to take further action? Like add some version information?

freeck, 2017-01-19:
Issue solved: see https://wiki.eclipse.org/CDT/User/NewIn92#Hide_gdb_traces_by_default

> To enable those traces go to Preferences->C/C++->Debug->GDB and enable the preference called "Show the GDB traces consoles". The size of the console buffer can still be set using the corresponding text preference. Note that the GDB traces will always be enabled and be recorded and that the preference in question is only used to display or hide the actual 'gdb traces' console.

This flag is now default not enabled opposed to previous releases....

closed 2017-01-19 by freeck