Sloeber/arduino-eclipse-plugin #639: Make "clean" problem in V4 windows: \ is replaced by /

ghislainborremans, 2017-01-21:
The del command contains file references that are correct ( with \ ) and a t a certain moment it changes to "/" making it a illegal switch of the del command and then it goes back to "\". 
see below, i left out parts (-----) for readability.

del   .\core\umm_malloc.c.d  .\core\spiffs_cache.c.d -------\core\uart.c.d  D:/Electronica/Workspace/serialtest/Release/arduino.ar  ./core/umm_malloc.c.o  ./core/spiffs_cache.c.o ./core/spiffs_check.c.o ./core/spiffs_gc.c.o ------- ./.ino.cpp.o  serialtest.hex  .\serialtest.ino.d  -------- .\core\spiffs_hal.cpp.d 
Invalid switch - "Electronica".
makefile:95: recipe for target 'clean' failed
make: [clean] Error 1 (ignored)

jantje, 2017-01-21:
Thanks for the input. I havn't noticed this yet. I'll try to reproduce.

jantje, 2017-01-23:
I have the same problem here. 
The del command should be os safe paths and quoted

ghislainborremans, 2017-01-23:
Ok, as said, the same happens also with the combiner ( i am old school, i suppose this is the linker?), resulting in "undefined references"

jantje, 2017-01-23:
Can you provide an example? 
We tested 222 boards on windows and a compiled lots of examples without problems.
We did not test clean.

ghislainborremans, 2017-01-24:
I am testing the startHere from painlessmesh. 

the problem arises when i add "includes" that refer to a place on the disk C:\sdfsdfsd or D:\sdfsdf. 
when i make a sketch without additional includes, all is ok. 
the problem arises both on windows and on linux. 
on windows the \ is changed to / 
on linux the path for the inluce is not quoted

ghislainborremans, 2017-01-28:
Hello,
is there a way to have this resolved as i can no longer use this , not on windows nor on linux? 

best regards

jantje, 2017-01-28:
I'm sorry. I wasn't aware you were blocked.
I have pushed a code change and started a build that should fix this in windows.
In about 2 hours there should be a nightly update available. Lets hope this fixes it in windows.

As to the linux problem. 

> on linux the path for the include is not quoted

Why is there a need for quotes? do you have spaces in the path?

jantje, 2017-01-28:
There are problems with the build server.
I'm on it.

ghislainborremans, 2017-01-28:
i will check on th linux version on monday, i thaugth there was a similar problem creating "undefined references". I have  no access to it today.

jantje, 2017-01-28:
Ok the new version is available.
I hope this is better :-)

jantje, 2017-01-28:
For your info. The only real clean I thrust is deleting the release folder

jantje, 2017-01-28:
I just tested it here. Sorry but a new build seems to be needed as this is release is not containing the changes I made

jantje, 2017-01-28:
The version is available now.

ghislainborremans, 2017-01-29:
sorry to bother you but where can i find the build? the nightly goes to 13/01/2017

jantje, 2017-01-29:
I used http://eclipse.baeyens.it/update/V4/nightly to upgrade to 28/01/2017

ghislainborremans, 2017-01-29:
The problem remains. I noticed that if i use the "project-settings-include" to indicate where source files resided, these sources are not used in teh combiner. 
When i add them as sources to the project , then it runs fine. So the problem lies in "including the src files"

jantje, 2017-01-29:
Are you still having / in the del command?

>I noticed that if i use the "project-settings-include" to indicate where source files resided, these sources are not used in teh combiner.

Are you talking about project properties->C/C++ general->Paths and symbols->tab includes?
There you should only refer to the include folders. This section is managed by Sloeber so making changes there may cause conflict.

Source files should always be added to the project. That is why core, variant and libraries are added to the project.

ghislainborremans, 2017-01-30:
The del command is OK 

i was referring to 
 project properties-> C/C++ Build -> Settings->Tool Settings->Arduino C++ Compiler-> Include Folders.

jantje, 2017-01-30:
>project properties-> C/C++ Build -> Settings->Tool Settings->Arduino C++ Compiler-> Include Folders.

Sloeber will not look for source code in the locations you add there.

If you want to add source code and not add it to the project (you can add by linking so no need to copy files around) you can add code in project properties-> C/C++ Build -> Settings->Tool Settings->Arduino C++ Compiler-> Include Folders.

jantje, 2017-01-30:
Just to summarize and if needed pick up remaining issues. Here is my current understanding of the situation:

1) The project clean issue on windows is resolved.
2) The linux problem is still open and needs some investigation on your side.

FYI are you aware there is a "add source folder to the selected project" in the arduino menu?
This wizard allows you to link in (no copy) code from any location on your hard drive. It sets the links and add the folder to the include path.
If you want a copy just drag and drop the code. Eclipse will ask whether you want a copy or a link. Select copy. If this code contains header files you will have to add the location to the include path.

ghislainborremans, 2017-01-30:
All ok now on windows and Linux.
Close issue.

jantje, 2017-01-30:
great :-)
Reopening as I close when the change is part of a release

closed 2017-05-25 by jantje