Sloeber/arduino-eclipse-plugin #643: Long folder names (with spaces) not correctly managed

EffegiWeb, 2017-01-26:
`14:14:28 **** Incremental Build of configuration Release for project RCX400Focuser ****
"D:\\sviluppo\\eclipse-test\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../RCX400Focuser.cpp'
'Starting C++ compile'
"D:\sviluppo\eclipse-test\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   -I"D:\sviluppo\eclipse-test\arduinoPlugin\packages\arduino\hardware\avr\1.6.17\cores\arduino" -I"D:\sviluppo\eclipse-test\arduinoPlugin\packages\arduino\hardware\avr\1.6.17\variants\mega" -MMD -MP -MF"RCX400Focuser.cpp.d" -MT"RCX400Focuser.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../RCX400Focuser.cpp"  -o  "RCX400Focuser.cpp.o"
'Finished building: ../RCX400Focuser.cpp'
' '
make: *** No rule to make target 'D:/work/Progetti/Focheggiatore', needed by 'RCX400Focuser.elf'.  Stop.

14:14:29 Build Finished (took 677ms)
`

real project folder Path is "D:\work\Progetti\Focheggiatore telescopio\Firmware foc\RCX400Focuser"

jantje, 2017-01-27:
Do not use spaces. They are known to cause problems.

EffegiWeb, 2017-01-27:
I'm actually use version 3 of Arduino plugin, with project in the same 
path Without any problem...

jantje, 2017-01-27:
Version 3 and 4 use a different makefile. With the makefile in V3 the spaces were handled ok but there were issues with sh and bash in the path being a problem as many people have cygwin or mingw or git in their path.
Also whether it works with spaces or not is very dependent on the platform.txt file provided with the board. This is out of my control. so in general: Better be safe than sorry : so do not use spaces.

EffegiWeb, 2017-01-30:
Thanks for the response: I've converted the project to V4 without using space in dir names.

jantje, 2017-01-30:
I have been looking in this issue more closely and basically make can not handle spaces.
There are tricks, but it remain tricks.

closed 2017-05-25 by jantje

wimjongman, 2017-07-13:
Is ANT and option?

jantje, 2017-07-13:
make is an integral part of the way cdt works.
Even if we found a make that supports windows mac and linux it still would be hard to get things to work properly because of the commands coming from platform.txt.
So basically sloeber does not fully control the command making it even harder to support spaces.

jantje, 2017-07-13:
Thinking about this. gnu arm eclipse seems to have found a "working solution" though they do not have the "platform.txt" problem