Sloeber/arduino-eclipse-plugin #644: V4, clean project gives error

hreintke, 2017-01-27:
Using Product V4, Windows 8.1

When doing a "clean project" you get the error 

```
"C:\\Utilities\\sloeber\\arduinoPlugin\\tools\\make\\make" clean
del   .\core\umm_malloc.c.d  .\core\spiffs_cache.c.d .\core\spiffs_check.c.d........
makefile:97: recipe for target 'clean' failed
process_begin: CreateProcess(NULL, del .\core\umm_malloc.c.d .\core\spiffs_cache.c.d.........
make (e=2): The system cannot find the file specified.

make: [clean] Error 2 (ignored)
```

jantje, 2017-01-27:
That is wierd.
Can you send me the generated release/makefile?

hreintke, 2017-01-28:
See the files below. Had to make the ".txt" to be able to attach.

Can you pint me to the sloeber location/files from which these files are generated 

[makefile.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/736832/makefile.txt)
[objects.mk.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/736833/objects.mk.txt)
[sources.mk.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/736835/sources.mk.txt)
[subdir.mk.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/736836/subdir.mk.txt)

jantje, 2017-01-28:
I see nothing strange in the generated files.
Can you run 
`D:\\embedded\\V4\\sloeber\\arduinoPlugin\\tools\\make\\make -version`

I get:
```
C:\Users\jan>D:\\embedded\\V4\\sloeber\\arduinoPlugin\\tools\\make\\make -version
GNU Make 3.82.90
Built for i686-pc-mingw32
Copyright (C) 1988-2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

C:\Users\jan>
```

hreintke, 2017-01-28:
I have the same version.

```
C:\Utilities\sloeber\arduinoPlugin\tools\make>make -version
GNU Make 3.82.90
Built for i686-pc-mingw32
Copyright (C) 1988-2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

jantje, 2017-01-28:
I'm wondering whether this would be 8.1 related.
I have pushed the changes I made to make the slashes OS dependent and forced a build.
Can you check with the new version (based on the nightly) in about 2 hours from here?

jantje, 2017-01-28:
that is issue #639

jantje, 2017-01-28:
Are there any  /p /f /s /q /a in the command?

hreintke, 2017-01-28:
Yes there are.

This is the complete command : 
```
del   .\core\umm_malloc.c.d  .\core\spiffs_cache.c.d .\core\spiffs_check.c.d .\core\spiffs_gc.c.d .\core\spiffs_hydrogen.c.d .\core\spiffs_nucleus.c.d  .\core\cdecode.c.d .\core\cencode.c.d  .\core\cont_util.c.d .\core\core_esp8266_eboot_command.c.d .\core\core_esp8266_flash_utils.c.d .\core\core_esp8266_i2s.c.d .\core\core_esp8266_noniso.c.d .\core\core_esp8266_phy.c.d .\core\core_esp8266_postmortem.c.d .\core\core_esp8266_si2c.c.d .\core\core_esp8266_timer.c.d .\core\core_esp8266_wiring.c.d .\core\core_esp8266_wiring_analog.c.d .\core\core_esp8266_wiring_digital.c.d .\core\core_esp8266_wiring_pulse.c.d .\core\core_esp8266_wiring_pwm.c.d .\core\core_esp8266_wiring_shift.c.d .\core\heap.c.d .\core\libc_replacements.c.d .\core\time.c.d .\core\uart.c.d  C:/Userdata/sloeber-workspace/teste/Release/arduino.ar  ./core/umm_malloc.c.o  ./core/spiffs_cache.c.o ./core/spiffs_check.c.o ./core/spiffs_gc.c.o ./core/spiffs_hydrogen.c.o ./core/spiffs_nucleus.c.o  ./core/cdecode.c.o ./core/cencode.c.o  ./core/Esp.cpp.o ./core/FS.cpp.o ./core/FunctionalInterrupt.cpp.o ./core/HardwareSerial.cpp.o ./core/IPAddress.cpp.o ./core/MD5Builder.cpp.o ./core/Print.cpp.o ./core/Schedule.cpp.o ./core/Stream.cpp.o ./core/StreamString.cpp.o ./core/Tone.cpp.o ./core/Updater.cpp.o ./core/WMath.cpp.o ./core/WString.cpp.o ./core/abi.cpp.o ./core/base64.cpp.o ./core/cbuf.cpp.o ./core/cont.S.o ./core/cont_util.c.o ./core/core_esp8266_eboot_command.c.o ./core/core_esp8266_flash_utils.c.o ./core/core_esp8266_i2s.c.o ./core/core_esp8266_main.cpp.o ./core/core_esp8266_noniso.c.o ./core/core_esp8266_phy.c.o ./core/core_esp8266_postmortem.c.o ./core/core_esp8266_si2c.c.o ./core/core_esp8266_timer.c.o ./core/core_esp8266_wiring.c.o ./core/core_esp8266_wiring_analog.c.o ./core/core_esp8266_wiring_digital.c.o ./core/core_esp8266_wiring_pulse.c.o ./core/core_esp8266_wiring_pwm.c.o ./core/core_esp8266_wiring_shift.c.o ./core/debug.cpp.o ./core/heap.c.o ./core/libc_replacements.c.o ./core/pgmspace.cpp.o ./core/spiffs_api.cpp.o ./core/spiffs_hal.cpp.o ./core/time.c.o ./core/uart.c.o  ./.ino.cpp.o  ./libraries/ESP8266mDNS/ESP8266mDNS.cpp.o  ./libraries/ESP8266WiFi/src/ESP8266WiFi.cpp.o ./libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp.o ./libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp.o ./libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp.o ./libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp.o ./libraries/ESP8266WiFi/src/ESP8266WiFiScan.cpp.o ./libraries/ESP8266WiFi/src/WiFiClient.cpp.o ./libraries/ESP8266WiFi/src/WiFiClientSecure.cpp.o ./libraries/ESP8266WiFi/src/WiFiServer.cpp.o ./libraries/ESP8266WiFi/src/WiFiUdp.cpp.o  teste.hex  .\teste.ino.d  .\core\cont.S.d  teste.elf  .\.ino.cpp.d  .\libraries\ESP8266mDNS\ESP8266mDNS.cpp.d  .\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp.d .\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp.d .\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp.d .\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp.d .\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp.d .\libraries\ESP8266WiFi\src\ESP8266WiFiScan.cpp.d .\libraries\ESP8266WiFi\src\WiFiClient.cpp.d .\libraries\ESP8266WiFi\src\WiFiClientSecure.cpp.d .\libraries\ESP8266WiFi\src\WiFiServer.cpp.d .\libraries\ESP8266WiFi\src\WiFiUdp.cpp.d  .\core\Esp.cpp.d .\core\FS.cpp.d .\core\FunctionalInterrupt.cpp.d .\core\HardwareSerial.cpp.d .\core\IPAddress.cpp.d .\core\MD5Builder.cpp.d .\core\Print.cpp.d .\core\Schedule.cpp.d .\core\Stream.cpp.d .\core\StreamString.cpp.d .\core\Tone.cpp.d .\core\Updater.cpp.d .\core\WMath.cpp.d .\core\WString.cpp.d .\core\abi.cpp.d .\core\base64.cpp.d .\core\cbuf.cpp.d .\core\core_esp8266_main.cpp.d .\core\debug.cpp.d .\core\pgmspace.cpp.d .\core\spiffs_api.cpp.d .\core\spiffs_hal.cpp.d 
```

jantje, 2017-01-28:
lets hope this is the problem and using \ will fix it.
I'm fighting my build server right now.

jantje, 2017-01-28:
for your info these chars are del commands
The fact that del is not a executable on your system together with these special instructions may explain the behaviour. (Though only testing will confirm)

hreintke, 2017-01-28:
OK,

I am checking this site for the new version, is that correct ?

`https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/4.0.0-SNAPSHOT/`

jantje, 2017-01-28:
I still need to update the website We are at 4.0.1

jantje, 2017-01-28:
You can download a completely new product there or -if all goes well-wait about a hour and do a upgrade.
The build server has restarted and the build is running.

jantje, 2017-01-28:
The version is available now.

hreintke, 2017-01-28:
Did a first quick check. I used the "update option".

Now compile fails.

```
Starting archiver
"C:\Utilities\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru "C:\Userdata\sloeber-workspace\teste/Release/arduino.ar"    C:/Userdata/sloeber-workspace/teste/Release/arduino.ar  .\core\umm_malloc.c.o  .\core\spiffs_cache.c.o .\core\spiffs_check.c.o .\core\spiffs_gc.c.o .\core\spiffs_hydrogen.c.o .\core\spiffs_nucleus.c.o  .\core\cdecode.c.o .\core\cencode.c.o  .\core\Esp.cpp.o .\core\FS.cpp.o .\core\FunctionalInterrupt.cpp.o .\core\HardwareSerial.cpp.o .\core\IPAddress.cpp.o .\core\MD5Builder.cpp.o .\core\Print.cpp.o .\core\Schedule.cpp.o .\core\Stream.cpp.o .\core\StreamString.cpp.o .\core\Tone.cpp.o .\core\Updater.cpp.o .\core\WMath.cpp.o .\core\WString.cpp.o .\core\abi.cpp.o .\core\base64.cpp.o .\core\cbuf.cpp.o .\core\cont.S.o .\core\cont_util.c.o .\core\core_esp8266_eboot_command.c.o .\core\core_esp8266_flash_utils.c.o .\core\core_esp8266_i2s.c.o .\core\core_esp8266_main.cpp.o .\core\core_esp8266_noniso.c.o .\core\core_esp8266_phy.c.o .\core\core_esp8266_postmortem.c.o .\core\core_esp8266_si2c.c.o .\core\core_esp8266_timer.c.o .\core\core_esp8266_wiring.c.o .\core\core_esp8266_wiring_analog.c.o .\core\core_esp8266_wiring_digital.c.o .\core\core_esp8266_wiring_pulse.c.o .\core\core_esp8266_wiring_pwm.c.o .\core\core_esp8266_wiring_shift.c.o .\core\debug.cpp.o .\core\heap.c.o .\core\libc_replacements.c.o .\core\pgmspace.cpp.o .\core\spiffs_api.cpp.o .\core\spiffs_hal.cpp.o .\core\time.c.o .\core\uart.c.o 
c:\Utilities\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\xtensa-lx106-elf-ar.exe: .coreumm_malloc.c.o: No such file or directory
makefile:78: recipe for target 'C:\Userdata\sloeber-workspace\teste\Release\arduino.ar' failed
make: *** [C:\Userdata\sloeber-workspace\teste\Release\arduino.ar] Error 1
```
See the ` .coreumm_malloc.c.o:` in there, maybe still an issue with / or \

But as said only first check, Will do some more tomorrow.
And if you want me to do some specific test, just let me know.

jantje, 2017-01-28:
did you delete the release folder before trying to build?

hreintke, 2017-01-29:
Yes I did.
Also tried both Nano and ESP8266 toolchains, same issue.

This is the Nano error message : 
```
Starting archiver
"C:\Utilities\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc-ar" rcs    C:/Userdata/sloeber-workspace/test1/Release/arduino.ar  .\core\CDC.cpp.o .\core\HardwareSerial.cpp.o .\core\HardwareSerial0.cpp.o .\core\HardwareSerial1.cpp.o .\core\HardwareSerial2.cpp.o .\core\HardwareSerial3.cpp.o .\core\IPAddress.cpp.o .\core\PluggableUSB.cpp.o .\core\Print.cpp.o .\core\Stream.cpp.o .\core\Tone.cpp.o .\core\USBCore.cpp.o .\core\WInterrupts.c.o .\core\WMath.cpp.o .\core\WString.cpp.o .\core\abi.cpp.o .\core\hooks.c.o .\core\main.cpp.o .\core\new.cpp.o .\core\wiring.c.o .\core\wiring_analog.c.o .\core\wiring_digital.c.o .\core\wiring_pulse.S.o .\core\wiring_pulse.c.o .\core\wiring_shift.c.o 
c:\utilities\sloeber\arduinoplugin\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: .coreCDC.cpp.o: No such file or directory
makefile:73: recipe for target 'C:\Userdata\sloeber-workspace\test1\Release\arduino.ar' failed
make: *** [C:\Userdata\sloeber-workspace\test1\Release\arduino.ar] Error 1
```

I want to do the tests also with a fresh, complete product install but cannot find that version.
Can you give me a link to that ?

jantje, 2017-01-29:
I asked you whether you removed the release folder because avr-gcc-ar complains about not finding coreumm_malloc.c.o while coreumm_malloc.c.o is not in the list of files to process.

In your second example this is exactly the same for coreCDC.cpp.o.

Looking closer I found that in your first example there is .\core\umm_malloc.c.o and in the second there is   .\core\CDC.cpp.o . **Both are the first file in the list.**
So basically on your system the \ is dropped.
The usual suspect for stuff like this is cygwin.
Do you have cygwin or something similar on your system?

jantje, 2017-01-29:
If you have cygwin on your system can you test starting sloeber with cygwin not in the path?

hreintke, 2017-01-29:
I do not have cygwin installed on my system.

Next test I did : 
In slober plugin start build.
Then I get the error.

opened cmd window.
cd to project\release dir
run the failing command -> no error
```
C:\Userdata\sloeber-workspace\teste\Release>del arduino.ar

C:\Userdata\sloeber-workspace\teste\Release>"C:\Utilities\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.
0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru "C:\Userdata\sloeber-workspace\teste/Release/arduino.ar"    C:/Userdata/sloeber-works
pace/teste/Release/arduino.ar  .\core\umm_malloc.c.o  .\core\spiffs_cache.c.o .\core\spiffs_check.c.o .\core\spiffs_gc.c.o .\core\
spiffs_hydrogen.c.o .\core\spiffs_nucleus.c.o  .\core\cdecode.c.o .\core\cencode.c.o  .\core\Esp.cpp.o .\core\FS.cpp.o .\core\Func
tionalInterrupt.cpp.o .\core\HardwareSerial.cpp.o .\core\IPAddress.cpp.o .\core\MD5Builder.cpp.o .\core\Print.cpp.o .\core\Schedul
e.cpp.o .\core\Stream.cpp.o .\core\StreamString.cpp.o .\core\Tone.cpp.o .\core\Updater.cpp.o .\core\WMath.cpp.o .\core\WString.cpp
.o .\core\abi.cpp.o .\core\base64.cpp.o .\core\cbuf.cpp.o .\core\cont.S.o .\core\cont_util.c.o .\core\core_esp8266_eboot_command.c
.o .\core\core_esp8266_flash_utils.c.o .\core\core_esp8266_i2s.c.o .\core\core_esp8266_main.cpp.o .\core\core_esp8266_noniso.c.o .
\core\core_esp8266_phy.c.o .\core\core_esp8266_postmortem.c.o .\core\core_esp8266_si2c.c.o .\core\core_esp8266_timer.c.o .\core\co
re_esp8266_wiring.c.o .\core\core_esp8266_wiring_analog.c.o .\core\core_esp8266_wiring_digital.c.o .\core\core_esp8266_wiring_puls
e.c.o .\core\core_esp8266_wiring_pwm.c.o .\core\core_esp8266_wiring_shift.c.o .\core\debug.cpp.o .\core\heap.c.o .\core\libc_repla
cements.c.o .\core\pgmspace.cpp.o .\core\spiffs_api.cpp.o .\core\spiffs_hal.cpp.o .\core\time.c.o .\core\uart.c.o

C:\Userdata\sloeber-workspace\teste\Release>dir
 Volume in drive C has no label.
 Volume Serial Number is 4AFA-EF19

 Directory of C:\Userdata\sloeber-workspace\teste\Release

29-01-2017  17:35    <DIR>          .
29-01-2017  17:35    <DIR>          ..
29-01-2017  17:26            32.307 .ino.cpp.d
29-01-2017  17:26            42.984 .ino.cpp.o
29-01-2017  17:35         2.005.106 arduino.ar
29-01-2017  17:27    <DIR>          core
29-01-2017  17:26    <DIR>          libraries
29-01-2017  17:26             4.211 makefile
29-01-2017  17:26               239 objects.mk
29-01-2017  17:26               835 sources.mk
29-01-2017  17:26             3.756 subdir.mk
               7 File(s)      2.089.438 bytes
               4 Dir(s)  95.928.737.792 bytes free

C:\Userdata\sloeber-workspace\teste\Release>cygwin
'cygwin' is not recognized as an internal or external command,
```

Will continue testing/experimenting on monday

jantje, 2017-01-29:
and if you run make there?

jantje, 2017-01-30:
I'd really like to know what is going on on your system. So please keep me posted.

hreintke, 2017-01-30:
Sure I will keep you posted. Did not have the time yet today to do the testing.
What command of make do you want me to start ?
Any specific version of sloeber to use ? 
When starting : 
`C:\Userdata\sloeber-workspace\teste\Release>c:\Utilities\sloeber\arduinoPlugin\tools\make\make.exe clean`
I get the original error
```
make (e=2): The system cannot find the file specified.
makefile:97: recipe for target 'clean' failed
make: [clean] Error 2 (ignored)
```

jantje, 2017-01-30:
Apparently you are the only one with this problem. At least all other issues in relation to windows clean are fixed with the nightly. 
When running make on the command line do you also get?

> process_begin: CreateProcess(NULL, del .\core\umm_malloc.c.d .\core\spiffs_cache.c.d...

hreintke, 2017-01-31:
Yes it does.
The complete lines are 
```
c:\Userdata\sloeber-workspace\teste\Release>c:\Utilities\sloeber\arduinoPlugin\tools\make\make.exe clean
del   .\core\umm_malloc.c.d  .\core\spiffs_cache.c.d .\core\spiffs_check.c.d .\core\spiffs_gc.c.d .\core\spiffs_hydrogen.c.d .\cor
e\spiffs_nucleus.c.d  .\core\cdecode.c.d .\core\cencode.c.d  .\core\cont_util.c.d .\core\core_esp8266_eboot_command.c.d .\core\cor
e_esp8266_flash_utils.c.d .\core\core_esp8266_i2s.c.d .\core\core_esp8266_noniso.c.d .\core\core_esp8266_phy.c.d .\core\core_esp82
66_postmortem.c.d .\core\core_esp8266_si2c.c.d .\core\core_esp8266_timer.c.d .\core\core_esp8266_wiring.c.d .\core\core_esp8266_wi
ring_analog.c.d .\core\core_esp8266_wiring_digital.c.d .\core\core_esp8266_wiring_pulse.c.d .\core\core_esp8266_wiring_pwm.c.d .\c
ore\core_esp8266_wiring_shift.c.d .\core\heap.c.d .\core\libc_replacements.c.d .\core\time.c.d .\core\uart.c.d  C:\Userdata\sloebe
r-workspace\teste\Release\arduino.ar  .\core\umm_malloc.c.o  .\core\spiffs_cache.c.o .\core\spiffs_check.c.o .\core\spiffs_gc.c.o
.\core\spiffs_hydrogen.c.o .\core\spiffs_nucleus.c.o  .\core\cdecode.c.o .\core\cencode.c.o  .\core\Esp.cpp.o .\core\FS.cpp.o .\co
re\FunctionalInterrupt.cpp.o .\core\HardwareSerial.cpp.o .\core\IPAddress.cpp.o .\core\MD5Builder.cpp.o .\core\Print.cpp.o .\core\
Schedule.cpp.o .\core\Stream.cpp.o .\core\StreamString.cpp.o .\core\Tone.cpp.o .\core\Updater.cpp.o .\core\WMath.cpp.o .\core\WStr
ing.cpp.o .\core\abi.cpp.o .\core\base64.cpp.o .\core\cbuf.cpp.o .\core\cont.S.o .\core\cont_util.c.o .\core\core_esp8266_eboot_co
mmand.c.o .\core\core_esp8266_flash_utils.c.o .\core\core_esp8266_i2s.c.o .\core\core_esp8266_main.cpp.o .\core\core_esp8266_nonis
o.c.o .\core\core_esp8266_phy.c.o .\core\core_esp8266_postmortem.c.o .\core\core_esp8266_si2c.c.o .\core\core_esp8266_timer.c.o .\
core\core_esp8266_wiring.c.o .\core\core_esp8266_wiring_analog.c.o .\core\core_esp8266_wiring_digital.c.o .\core\core_esp8266_wiri
ng_pulse.c.o .\core\core_esp8266_wiring_pwm.c.o .\core\core_esp8266_wiring_shift.c.o .\core\debug.cpp.o .\core\heap.c.o .\core\lib
c_replacements.c.o .\core\pgmspace.cpp.o .\core\spiffs_api.cpp.o .\core\spiffs_hal.cpp.o .\core\time.c.o .\core\uart.c.o  .\.ino.c
pp.o  .\libraries\ESP8266mDNS\ESP8266mDNS.cpp.o  .\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp.o .\libraries\ESP8266WiFi\src\ESP8266
WiFiAP.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp.o .\libraries\E
SP8266WiFi\src\ESP8266WiFiSTA.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiScan.cpp.o .\libraries\ESP8266WiFi\src\WiFiClient.cpp.o
 .\libraries\ESP8266WiFi\src\WiFiClientSecure.cpp.o .\libraries\ESP8266WiFi\src\WiFiServer.cpp.o .\libraries\ESP8266WiFi\src\WiFiU
dp.cpp.o  teste.hex  .\teste.ino.d  .\core\cont.S.d  teste.elf  .\.ino.cpp.d  .\libraries\ESP8266mDNS\ESP8266mDNS.cpp.d  .\librari
es\ESP8266WiFi\src\ESP8266WiFi.cpp.d .\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp.d .\libraries\ESP8266WiFi\src\ESP8266WiFiGeneri
c.cpp.d .\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp.d .\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp.d .\libraries\ESP8266WiF
i\src\ESP8266WiFiScan.cpp.d .\libraries\ESP8266WiFi\src\WiFiClient.cpp.d .\libraries\ESP8266WiFi\src\WiFiClientSecure.cpp.d .\libr
aries\ESP8266WiFi\src\WiFiServer.cpp.d .\libraries\ESP8266WiFi\src\WiFiUdp.cpp.d  .\core\Esp.cpp.d .\core\FS.cpp.d .\core\Function
alInterrupt.cpp.d .\core\HardwareSerial.cpp.d .\core\IPAddress.cpp.d .\core\MD5Builder.cpp.d .\core\Print.cpp.d .\core\Schedule.cp
p.d .\core\Stream.cpp.d .\core\StreamString.cpp.d .\core\Tone.cpp.d .\core\Updater.cpp.d .\core\WMath.cpp.d .\core\WString.cpp.d .
\core\abi.cpp.d .\core\base64.cpp.d .\core\cbuf.cpp.d .\core\core_esp8266_main.cpp.d .\core\debug.cpp.d .\core\pgmspace.cpp.d .\co
re\spiffs_api.cpp.d .\core\spiffs_hal.cpp.d
process_begin: CreateProcess(NULL, del .\core\umm_malloc.c.d .\core\spiffs_cache.c.d .\core\spiffs_check.c.d .\core\spiffs_gc.c.d
.\core\spiffs_hydrogen.c.d .\core\spiffs_nucleus.c.d .\core\cdecode.c.d .\core\cencode.c.d .\core\cont_util.c.d .\core\core_esp826
6_eboot_command.c.d .\core\core_esp8266_flash_utils.c.d .\core\core_esp8266_i2s.c.d .\core\core_esp8266_noniso.c.d .\core\core_esp
8266_phy.c.d .\core\core_esp8266_postmortem.c.d .\core\core_esp8266_si2c.c.d .\core\core_esp8266_timer.c.d .\core\core_esp8266_wir
ing.c.d .\core\core_esp8266_wiring_analog.c.d .\core\core_esp8266_wiring_digital.c.d .\core\core_esp8266_wiring_pulse.c.d .\core\c
ore_esp8266_wiring_pwm.c.d .\core\core_esp8266_wiring_shift.c.d .\core\heap.c.d .\core\libc_replacements.c.d .\core\time.c.d .\cor
e\uart.c.d C:\Userdata\sloeber-workspace\teste\Release\arduino.ar .\core\umm_malloc.c.o .\core\spiffs_cache.c.o .\core\spiffs_chec
k.c.o .\core\spiffs_gc.c.o .\core\spiffs_hydrogen.c.o .\core\spiffs_nucleus.c.o .\core\cdecode.c.o .\core\cencode.c.o .\core\Esp.c
pp.o .\core\FS.cpp.o .\core\FunctionalInterrupt.cpp.o .\core\HardwareSerial.cpp.o .\core\IPAddress.cpp.o .\core\MD5Builder.cpp.o .
\core\Print.cpp.o .\core\Schedule.cpp.o .\core\Stream.cpp.o .\core\StreamString.cpp.o .\core\Tone.cpp.o .\core\Updater.cpp.o .\cor
e\WMath.cpp.o .\core\WString.cpp.o .\core\abi.cpp.o .\core\base64.cpp.o .\core\cbuf.cpp.o .\core\cont.S.o .\core\cont_util.c.o .\c
ore\core_esp8266_eboot_command.c.o .\core\core_esp8266_flash_utils.c.o .\core\core_esp8266_i2s.c.o .\core\core_esp8266_main.cpp.o
[... cut ...]
re\Schedule.cpp.d .\core\Stream.cpp.d .\core\StreamString.cpp.d .\core\Tone.cpp.d .\core\Updater.cpp.d .\core\WMath.cpp.d .\core\W
String.cpp.d .\core\abi.cpp.d .\core\base64.cpp.d .\core\cbuf.cpp.d .\core\core_esp8266_main.cpp.d .\core\debug.cpp.d .\core\pgmsp
ace.cpp.d .\core\spiffs_api.cpp.d .\core\spiffs_hal.cpp.d, ...) failed.
make (e=2): The system cannot find the file specified.
makefile:97: recipe for target 'clean' failed
make: [clean] Error 2 (ignored)
```

Maybe/probably not related but I remember to have seen a message that the command del was not recognized. Cannot remember when and cannot reproduce.

jantje, 2017-01-31:
this is really strange.
Sloeber generates the makefiles and gives CDT the make location.
CDT runs make.
But make run from eclipse gives problems
make run from the command line does not.

jantje, 2017-01-31:
can you run run make from the external commands inside eclipse.

hreintke, 2017-01-31:
On your remark : 
>But make run from eclipse gives problems
make run from the command line does not

Does it mean it is reproducible in your config ? 

See the attached screenshot, is that what you want to see ?

[image]

jantje, 2017-02-01:
Indeed, this is what I wanted to see.
As the c:\Utilities\sloeber\arduinoPlugin\tools\make\make.exe clean runs from the command line but not from the eclipse console there must be a difference between the 2.
Do you start Sloeber with a command that changes the path?
Do you run a 64 or 32 bit OS?
Do you run 64 or 32 bit java?

And no I can not reproduce the issue and there is no open question/issue that is in any way similar.

hreintke, 2017-02-01:
I don't understand your statement on `runs from the command line but not from the eclipse console`
Looks to me that they both give the same error.

On your questions : 
I run sloeber from windows-explorer, no changes to the path.
Windows 8.1 64 bit
Java 64 bit.

I would like to make a complete new product install but on the link `https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/4.0.0-SNAPSHOT/` there is no recent nightly version. Any other location I should use ?

jantje, 2017-02-01:
>I don't understand your statement on runs from the command line but not from the eclipse console
Looks to me that they both give the same error.

You are right, my bad.
so make clean does not work in eclipse and does not work on the command line.
Given the fact that the make is the correct make and that nothing seems wrong in the makefiles I don't see how a new install would fix it. But it is something you can try.
There has not been a new nightly since the \ / release.

jantje, 2017-02-01:
The nightly has upgraded version number so it is
https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/4.0.1-SNAPSHOT/

hreintke, 2017-02-01:
Thx,
Need/will do install from that. Will not be today.
Let you know the results.

jantje, 2017-02-01:
Been reading up on this and have you already tried to create a del.bat?

jantje, 2017-02-01:
I think this is related to windows 8.1 and the cmd /c del.
More to be found here https://bugs.eclipse.org/bugs/show_bug.cgi?id=242649

hreintke, 2017-02-01:
Yes, might well be. Error message is similar to what I get.

What should be in the del.bat ?
Can I update sloeber to use cmd /c del ? Alltough I expect that filenames have to be quoted.

jantje, 2017-02-01:
>Can I update sloeber to use cmd /c del ?

I already cheked that. The del is hard coded.
I'm currently installing mingw64 to see how that works.
Might be an alternative for you

jantje, 2017-02-01:
Ran my project here clean and buid with the make64 32 bit make
Worked fine
this is its info
```
C:\Users\jan>D:\embedded\eclipse\arduinoPlugin\tools\make\make.exe -version
GNU Make 4.1
Built for x86_64-w64-mingw32
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```
You can download the make directly from
http://eclipse.baeyens.it/download/make.exe

hreintke, 2017-02-01:
Downloaded your make.
I am sure I am running that one and still have the error.

But.. Having been so much messing around with versions, I reverted to the original 4.0 version.
Can't work on it now, but will upgrade tomorrow and test with 4.0.1

BTW : Thanks for putting all the effort in this

jantje, 2017-02-01:
I will not have much time in the coming 10 days to help you out. But I'll try to follow up.
In the del.bat put echo deleting. So we know it is finding the file.
Then if this works simply del * should do

hreintke, 2017-02-02:
No problem that you have not much time.
I easily can just delete the release folder manually to clean the project.

Did a first try with del.bat, got error
```
Invalid switch - "Userdata".
make: [clean] Error 1 (ignored)
makefile:102: recipe for target 'clean' failed
```
Will dig into that.

BTW : I downloaded/compiled sloeber from git, maybe gives flexibility for quick checking
Compiles OK (no errors) but running win32x64 gives the  error : 
```
The path C:/temp/arduino-eclipse-plugin/io.sloeber.product/target/products/io.sloeber.product/win32/win32/x86_64/sloeber/arduinoPlugin is to long
```
Did you see this before ?

jantje, 2017-02-02:
Yes I saw this before.
I coded that ;-).
Basically windows still has a path length limitation and packages can be really long.
https://github.com/Sloeber/arduino-eclipse-plugin/blob/0b368ffc38ebd52c1d3ee56b3c5ff4f61cc5e26d/io.sloeber.core/src/io/sloeber/core/Activator.java#L230
I prefer giving a warning to having unexplainable behaviour all the time.

As to the del.bat. I'm not sure what you did exactly. 
I would first test with a empty del.bat and the makefile on the command line.
If that does not generate a error I would try to put del * in the del.bat file.
if that works I would try in sloeber.

Are you still having the link error or did that go away? If it went away, do you know what made it go away?

jantje, 2017-02-12:
Are you still having the link error or did that go away? If it went away, do you know what made it go away?

hreintke, 2017-02-12:
Hi Jantje,

No it did not go away yet. I have an environment ready for further testing.
But I am (unexpected) very busy due to business. 
And I did not want to just to drop in the remark "it still doesn't work". 

Expect this to last until next weekend,

jantje, 2017-02-12:
don't worry.

jantje, 2017-03-13:
Any progress?

Normalised, 2017-03-14:
I've got the error building a clean project on windows.
I tried the nightly plugin : 4.0.1.201703140109
Created new Sketch, hit verify :

`15:26:51 **** Incremental Build of configuration Release for project TestArduino ****
"C:\\dev\\apps\\eclipse\\cpp-neon\\eclipse\\arduinoPlugin\\tools\\make\\make" all 
subdir.mk:24: recipe for target '.ino.cpp.o' failed
process_begin: CreateProcess(NULL, echo "Building file: ..\.ino.cpp", ...) failed.
make (e=2): The system cannot find the file specified.

make: *** [.ino.cpp.o] Error 2

15:26:51 Build Finished (took 235ms)
`

[makefile.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/841604/makefile.txt)

hreintke, 2017-03-14:
Hi,
Will pick it up. Was "distracted" by my esp project.

hreintke, 2017-03-15:
I am now using Windows 8.1 64 bit, Sloeber plugin : Git master.
Full clean install, also new workspace.
All directorynames are without blanks

New ESP project : 
```
Starting archiver
"C:\temp\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru "C:/temp/sw1/t4/Release/arduino.ar"    C:/temp/sw1/t4/Release/arduino.ar  .\core\umm_malloc.c.o  .\core\spiffs_cache.c.o .\core\spiffs_check.c.o .\core\spiffs_gc.c.o .\core\spiffs_hydrogen.c.o .\core\spiffs_nucleus.c.o  .\core\cdecode.c.o .\core\cencode.c.o  .\core\Esp.cpp.o .\core\FS.cpp.o .\core\HardwareSerial.cpp.o .\core\IPAddress.cpp.o .\core\MD5Builder.cpp.o .\core\Print.cpp.o .\core\Schedule.cpp.o .\core\Stream.cpp.o .\core\StreamString.cpp.o .\core\Tone.cpp.o .\core\Updater.cpp.o .\core\WMath.cpp.o .\core\WString.cpp.o .\core\abi.cpp.o .\core\base64.cpp.o .\core\cbuf.cpp.o .\core\cont.S.o .\core\cont_util.c.o .\core\core_esp8266_eboot_command.c.o .\core\core_esp8266_flash_utils.c.o .\core\core_esp8266_i2s.c.o .\core\core_esp8266_main.cpp.o .\core\core_esp8266_noniso.c.o .\core\core_esp8266_phy.c.o .\core\core_esp8266_postmortem.c.o .\core\core_esp8266_si2c.c.o .\core\core_esp8266_timer.c.o .\core\core_esp8266_wiring.c.o .\core\core_esp8266_wiring_analog.c.o .\core\core_esp8266_wiring_digital.c.o .\core\core_esp8266_wiring_pulse.c.o .\core\core_esp8266_wiring_pwm.c.o .\core\core_esp8266_wiring_shift.c.o .\core\debug.cpp.o .\core\heap.c.o .\core\libc_replacements.c.o .\core\pgmspace.cpp.o .\core\spiffs_api.cpp.o .\core\spiffs_hal.cpp.o .\core\time.c.o .\core\uart.c.o 
c:\temp\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\xtensa-lx106-elf-ar.exe: .coreumm_malloc.c.o: No such file or directory
makefile:76: recipe for target 'C:\temp\sw1\t4\Release\arduino.ar' failed
make: *** [C:\temp\sw1\t4\Release\arduino.ar] Error 1

17:39:06 Build Finished (took 32s.853ms)
```
New Arduino project

```
Starting archiver
"C:\temp\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc-ar" rcs    C:/temp/sw1/t2/Release/arduino.ar  .\core\CDC.cpp.o .\core\HardwareSerial.cpp.o .\core\HardwareSerial0.cpp.o .\core\HardwareSerial1.cpp.o .\core\HardwareSerial2.cpp.o .\core\HardwareSerial3.cpp.o .\core\IPAddress.cpp.o .\core\PluggableUSB.cpp.o .\core\Print.cpp.o .\core\Stream.cpp.o .\core\Tone.cpp.o .\core\USBCore.cpp.o .\core\WInterrupts.c.o .\core\WMath.cpp.o .\core\WString.cpp.o .\core\abi.cpp.o .\core\hooks.c.o .\core\main.cpp.o .\core\new.cpp.o .\core\wiring.c.o .\core\wiring_analog.c.o .\core\wiring_digital.c.o .\core\wiring_pulse.S.o .\core\wiring_pulse.c.o .\core\wiring_shift.c.o 
c:\temp\sloeber\arduinoplugin\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: .coreCDC.cpp.o: No such file or directory
makefile:73: recipe for target 'C:\temp\sw1\t2\Release\arduino.ar' failed
make: *** [C:\temp\sw1\t2\Release\arduino.ar] Error 1

17:40:13 Build Finished (took 14s.469ms)
```
When I start the verify again on the project I get : 
```
17:42:22 **** Incremental Build of configuration Release for project t2 ****
"C:\\temp\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
Starting combiner
"C:\temp\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o "C:/temp/sw1/t2/Release/t2.elf"    .\.ino.cpp.o   C:\temp\sw1\t2\Release\arduino.ar    C:/temp/sw1/t2/Release/arduino.ar  "-LC:/temp/sw1/t2/Release" -lm
avr-gcc.exe: error: ..ino.cpp.o: No such file or directory
avr-gcc.exe: error: C:tempsw1t2Releasearduino.ar: No such file or directory
makefile:79: recipe for target 't2.elf' failed
make: *** [t2.elf] Error 1

17:42:22 Build Finished (took 281ms)
```

jantje, 2017-03-15:
I see C:\temp\sw1\t2\Release\arduino.ar    C:/temp/sw1/t2/Release/arduino.ar  
so the archive is there with \ and /
I'll try to reproduce this here
Though I still don't see why your os has a problem with it :-(

jantje, 2017-03-15:
I can't reproduce this problem.
I did find this which may be related http://stackoverflow.com/questions/7250130/how-to-stop-mingw-and-msys-from-mangling-path-names-given-at-the-command-line/34386471#34386471

Normalised, 2017-03-15:
For info I don't have cygwin or mingw installed.

jantje, 2017-03-15:
make that is installed with sloeber is from mingw.
Do you have git stuff installed?

Normalised, 2017-03-15:
Ok, that is exactly what is causing the problem, if i remove Git from my path then the project builds without issue.

Of course I don't want to remove git from my path :)

jantje, 2017-03-15:
What do you mean with git?

Normalised, 2017-03-15:
I have the git binaries installed at C:\Program Files\Git which had added C:\Program Files\Git\bin to my system PATH. If I remove that from the PATH then Sloeber works fine.

jantje, 2017-03-16:
I installed this git just now. 
C:\Program Files\Git\cmd was in the path but the bin folder was not.
Tried sloeber, no problems.
Added the bin to the path as well. No problems here.
I used the default settings so maybe a special setting is needed.

Can you try setting MSYS_NO_PATHCONV=1 and MSYS2_ARG_CONV_EXCL="*" as environment variables and see whether that fixes the problem as well?

jantje, 2017-03-16:
I use git in eclipse and https://desktop.github.com/
Works fine and is not in the path

alim-s-ajiev, 2017-03-16:
Hello

I am trying to use Sloeber V4 in Windows 8.1. I have installed full Sloeber IDE and created a ESP8266 project. I have the following troubles:

1. When I make clean, it tries to execute windows "del" command with paths containing "/" character. This does not works because Windows believes that everything starting from "/" is a command option. So I have to delete Release folder manually for clean.

2. I need just 2 libraries for ESP8266: Blynk and ESP8266WiFi. The project builds successfully with just these 2 libraries. But IDE also includes some other libraries (Ethernet, WiFi, SD...) in my project which makes the project fail to build because these libraries also need AVR32 libraries and so on. Sometimes and somehow I succeed in removing these libraries, and the project is built successfully. But it is very tricky.

How can I overcome these problems?

Thanks for any reply!

hreintke, 2017-03-16:
Hi,

I removed git from the path. -> Same behavior as before

[12 more replies]

closed 2017-05-25 by jantje