Sloeber/arduino-eclipse-plugin #645: add support for ssh upload with yun

jantje, 2017-01-28:
Currently only upload via username and password is supported.

jantje, 2017-03-05:
Also a bug makes it impossible to change/add a username/password.

jantje, 2017-03-06:
Note that when going to [yun].local->login->configure->advanced configuration panel->system administration the ssh key should be on 1 line starting with the encryption type followed with 1 space. which is in my case ssh-rsa

[image]

jantje, 2017-03-06:
Setting/deleting passwords can be done in project properties->arduino
[image]
Which gives you this page
[image]
note the page will only show up when the port is a "accepted" machine name

mjmeijer, 2017-03-06:
Shouldn't that password dialog be a real password dialog, using SWT.PASSWORD as per https://goo.gl/vUz3Mn ??

jantje, 2017-03-06:
@mjmeijer 
I assume you propose to use a password edit box (one that shows * instead of the letters)
I prefer to see the password (I'm dyslectic) but a checkbox "show password" suits both needs.
If you want it secure; you should go for public private key. 
What I'm trying to say: "I can't be bothered. But if you can ... be my guest."

closed 2017-05-25 by jantje