Sloeber/arduino-eclipse-plugin #647: How to debug FreeRTOS

shineviayan, 2017-01-28:
I want to use Sloeber to compile and debug FreeRTOS Task on Arduino Due.
Is this supported ? All pointers are welcome

Thanks in advance
-Shine

jantje, 2017-01-28:
As far as I know the due has no native feature to debug.
So this would require a debugger.

freeck, 2017-09-13:
I have adopted a very pragmatic solution for in my case debugging ArduinoMini boards (this board only supports SWD- instead of JTAG-interface):
- buy an AtmelICE module (or any other compatible ICE for your processor supported by AtmelStudio)
- edit/compile/build with Sloeber (C/C++ build option set at -ggdb -o0)
- thereby creating a "elf" file
- start AtmelStudio, open the objectfile you just created
- start debugging!

jantje, 2017-12-11:
@freeck 
There must be a better way. But lets discuss during the esp32 debug I want to start early next year

closed 2017-12-11 by jantje

freeck, 2017-12-12:
That's what I though , but the SWD-protocol seems to be proprietary....