Sloeber/arduino-eclipse-plugin #648: V4 Upgrade mylyn and CDT fail

jantje, 2017-01-28:
I have installed V4.0 upgraded to the nightly and when I run update The updater proposes to update the mylyn and CDT plugins.
But he can not do so because conflicting dependencies.
So I tried CDT only but it failed again with this error

```
Cannot complete the install because of a conflicting dependency.
  Software being installed: C/C++ Development Platform 9.2.0.201612061315 (org.eclipse.cdt.platform.feature.group 9.2.0.201612061315)
  Software currently installed: Sloeber, the Eclipse Arduino IDE 4.0.0.201610300211 (io.sloeber.product 4.0.0.201610300211)
  Only one of the following can be installed at once: 
    org.eclipse.cdt.core.win32 5.4.1.201609121658
    org.eclipse.cdt.core.win32 5.4.1.201606062011
    org.eclipse.cdt.core.win32 5.4.1.201612061315
  Cannot satisfy dependency:
    From: Sloeber, the Eclipse Arduino IDE 4.0.0.201610300211 (io.sloeber.product 4.0.0.201610300211)
    To: org.eclipse.cdt.native.feature.group [9.1.0.201609121658]
  Cannot satisfy dependency:
    From: C/C++ Development Tooling Native Utilities 9.1.0.201609121658 (org.eclipse.cdt.native.feature.group 9.1.0.201609121658)
    To: org.eclipse.cdt.core.win32 [5.4.1.201609121658]
  Cannot satisfy dependency:
    From: C/C++ Development Tooling Native Utilities 9.2.0.201612061315 (org.eclipse.cdt.native.feature.group 9.2.0.201612061315)
    To: org.eclipse.cdt.core.win32 [5.4.1.201612061315]
  Cannot satisfy dependency:
    From: C/C++ Development Platform 9.2.0.201612061315 (org.eclipse.cdt.platform.feature.group 9.2.0.201612061315)
    To: org.eclipse.cdt.native.feature.group [9.2.0.201612061315]
```

Help is welcomed :-)

wimjongman, 2017-02-26:
What update sites do you have in your configuration?

jantje, 2017-02-26:
@wimjongman 
The problem seems to be with http://download.eclipse.org/tools/cdt/releases/9.1
A similar issue has been reported here https://github.com/Sloeber/arduino-eclipse-plugin/issues/665

closed 2017-05-25 by jantje