Sloeber/arduino-eclipse-plugin #649: V4 ESP8266 cannot create new Sketch (Finish Button)

darkhedie, 2017-01-31:
Hello

I just downloaded the latest Sloeber V4 IDE.
I have added the ESP8266 Packager with window->preferences -> arduino -> platforms and boards -> esp8266 -> esp8266 -> 2.3.0 

It was downloaded without problems. 

After that it tried to create a new sketch. 
Unfortunately, i am not be able to click the "finish" button. 
The button is grayed out. 
If i chose the arduino (atmel) from the dropdown list as platform folder, then i am able to click finish. 

System: 
Win10 x64
Latest Java (8 update 121)
[image]

jantje, 2017-01-31:
seems like the resize didn't work for you

closed 2017-03-06 by jantje

tischmi, 2017-03-12:
it's Works!!!

Gizmodo, 2019-07-31:
Same problem with window resize 
[image]