Sloeber/arduino-eclipse-plugin #658: error on launch

torbenw1996, 2017-02-09:
Hey,

i downloaded Sloeber on my pc and started to love it.
Now i downloaded it on my laptop and get the following error.

!SESSION 2017-02-09 09:02:12.380 -----------------------------------------------
eclipse.buildId=unknown
java.version=1.8.0_121
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -perspective io.sloeber.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective

!ENTRY org.eclipse.equinox.simpleconfigurator 4 0 2017-02-09 09:02:18.310
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Exception in org.eclipse.equinox.internal.simpleconfigurator.Activator.start() of bundle org.eclipse.equinox.simpleconfigurator.
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:795)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.start(BundleContextImpl.java:724)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.startWorker0(EquinoxBundle.java:932)
	at org.eclipse.osgi.internal.framework.EquinoxBundle$EquinoxModule.startWorker(EquinoxBundle.java:309)
	at org.eclipse.osgi.container.Module.doStart(Module.java:581)
	at org.eclipse.osgi.container.Module.start(Module.java:449)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1620)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1600)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1571)
	at org.eclipse.osgi.container.SystemModule.startWorker(SystemModule.java:264)
	at org.eclipse.osgi.container.Module.doStart(Module.java:581)
	at org.eclipse.osgi.container.Module.start(Module.java:449)
	at org.eclipse.osgi.container.SystemModule.start(SystemModule.java:188)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.start(EquinoxBundle.java:383)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.start(EquinoxBundle.java:402)
	at org.eclipse.osgi.launch.Equinox.start(Equinox.java:115)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.startup(EclipseStarter.java:326)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:239)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
Caused by: java.lang.IllegalArgumentException
	at org.eclipse.osgi.internal.framework.legacy.PackageAdminImpl.getBundles(PackageAdminImpl.java:172)
	at org.eclipse.equinox.internal.simpleconfigurator.ConfigApplier.refreshPackages(ConfigApplier.java:401)
	at org.eclipse.equinox.internal.simpleconfigurator.ConfigApplier.install(ConfigApplier.java:105)
	at org.eclipse.equinox.internal.simpleconfigurator.SimpleConfiguratorImpl.applyConfiguration(SimpleConfiguratorImpl.java:191)
	at org.eclipse.equinox.internal.simpleconfigurator.SimpleConfiguratorImpl.applyConfiguration(SimpleConfiguratorImpl.java:205)
	at org.eclipse.equinox.internal.simpleconfigurator.Activator.start(Activator.java:60)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:774)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:1)
	at java.security.AccessController.doPrivileged(Native Method)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:767)
	... 24 more
Root exception:
java.lang.IllegalArgumentException
	at org.eclipse.osgi.internal.framework.legacy.PackageAdminImpl.getBundles(PackageAdminImpl.java:172)
	at org.eclipse.equinox.internal.simpleconfigurator.ConfigApplier.refreshPackages(ConfigApplier.java:401)
	at org.eclipse.equinox.internal.simpleconfigurator.ConfigApplier.install(ConfigApplier.java:105)
	at org.eclipse.equinox.internal.simpleconfigurator.SimpleConfiguratorImpl.applyConfiguration(SimpleConfiguratorImpl.java:191)
	at org.eclipse.equinox.internal.simpleconfigurator.SimpleConfiguratorImpl.applyConfiguration(SimpleConfiguratorImpl.java:205)
	at org.eclipse.equinox.internal.simpleconfigurator.Activator.start(Activator.java:60)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:774)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:1)
	at java.security.AccessController.doPrivileged(Native Method)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:767)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.start(BundleContextImpl.java:724)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.startWorker0(EquinoxBundle.java:932)
	at org.eclipse.osgi.internal.framework.EquinoxBundle$EquinoxModule.startWorker(EquinoxBundle.java:309)
	at org.eclipse.osgi.container.Module.doStart(Module.java:581)
	at org.eclipse.osgi.container.Module.start(Module.java:449)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1620)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1600)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1571)
	at org.eclipse.osgi.container.SystemModule.startWorker(SystemModule.java:264)
	at org.eclipse.osgi.container.Module.doStart(Module.java:581)
	at org.eclipse.osgi.container.Module.start(Module.java:449)
	at org.eclipse.osgi.container.SystemModule.start(SystemModule.java:188)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.start(EquinoxBundle.java:383)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.start(EquinoxBundle.java:402)
	at org.eclipse.osgi.launch.Equinox.start(Equinox.java:115)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.startup(EclipseStarter.java:326)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:239)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)

!ENTRY org.eclipse.osgi 4 0 2017-02-09 09:02:18.320
!MESSAGE Bundle org.eclipse.equinox.simpleconfigurator_1.1.200.v20160504-1450 [5] is not active.

!ENTRY org.eclipse.osgi 4 0 2017-02-09 09:02:18.666
!MESSAGE Application error
!STACK 1
java.lang.IllegalStateException: Unable to acquire application service. Ensure that the org.eclipse.core.runtime bundle is resolved and started (see config.ini).
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:78)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)

pls help me

jantje, 2017-02-12:
did you install the product or the plugin?
Which version?

torbenw1996, 2017-02-12:
Product bundle 64bits V4.0

jantje, 2017-02-12:
stable?

torbenw1996, 2017-02-12:
yes
http://eclipse.baeyens.it/stable.php?OS=Windows
this 64bits bundle

jantje, 2017-02-12:
From what I see there seems to be a problem with the config.ini on your system.
Can you delete the whole sloeber installation and unzip again?

torbenw1996, 2017-02-12:
i tried this already three times :/

jantje, 2017-02-12:
have you compared the content of the sloeber-ide.ini and configuration/config.ini with what is on your pc? 
Have you tried to install and run other eclipse 64 bit products on your laptop?
I mean there is no sloeber in the trace so it looks like the problem is not sloeber related but more a OS/java/eclipse problem

wimjongman, 2017-02-26:
What is you install path?

jantje, 2017-03-06:
Is there an update on this

torbenw1996, 2017-03-07:
I dont got the prebuild pack running, a Clean Eclipse install and Plugin Installation worked fine

jantje, 2017-03-07:
txs for the feedback.
I'll try to install a 32 bit product to see whether that causes issue's on my system.

jantje, 2017-03-19:
I guess we can close this one

closed 2017-03-19 by jantje