Sloeber/arduino-eclipse-plugin #659: 4.0 on Mac Pro Yosemite fails to compile

MarkAllread, 2017-02-10:
Installed V4.0 on Mac. No issues occurred during installation.
Opened Sloeber and assigned current workspace.

-First thing I noticed was that my custom Perspective was unusable. (The default Perspective worked fine)
- Tried to compile and received the alert about upgrading projects. Went through the process with three projects and none of them would compile any more, due to a make error. Unfortunately I did not  take proper note of the issue as I quickly decided to go back to V3.1
- Going back to 3.1 was VERY difficult as I had to search for and destroy several files before I could get 3.1 to work again.

Fortunately I keep good backups so was eventually able to revert back, but not without a lot of trouble.

Next time I will create a clone drive to try V4

I'd like to take this opportunity to say thank you for Slobber. I'm just a casual hobby programmer but having LOADS of fun with it.

System:
2010 Mac Pro
16GB
OSX Yosemite 10.10.5

jantje, 2017-02-11:
Not sure why you created a issue.
Here is a video explaining how to do the upgrade: https://www.youtube.com/watch?v=p4M5itvX26g

MarkAllread, 2017-02-11:
Hi there.
Thanks for the video.
I created the issue because I followed the instructions that popped up whenI installed 4.0 
I tried to build a project from my 3.1 workspace. Just as in your video
I tried this three times with different projects.
Each time I completed the instructions I tried to build again. It did not give me the pop up, but the build failed each time with a make error.
I know I should have taken more careful note of the build error, but I did not, however all the projects would build just fine in 3.1.
Until I create a complete clone drive to retry this I can't provide further info. I am now back to 3.1 and working just fine.

jantje, 2017-02-12:
Thanks for the info.

jantje, 2017-03-06:
Should this issue remain open?

MarkAllread, 2017-03-07:
Somehow my last comment from 2 weeks ago didn't appear...
I created a clone drive and had the same issue.
I noted the error and thought I'd posted the details here but obvioulsy messed up while doing that.
I will grab the late beta and try again.

closed 2017-03-07 by MarkAllread