Sloeber/arduino-eclipse-plugin #660: Digispark asm code does not compile

ligius-, 2017-02-16:
Using V3 from the marketplace.
I'm trying to compile the Digispark examples which provide USB support, via some assembly instructions. For example the DigisparkJoystick library.

The first issue is that usbdrvasm.asm file has an "end" instruction at the end which is somehow not recognized.

The second issue is that inside the usbdrvasm.S file the #if blocks are not recognized (preprocessor is not working).

Even after the two issues above are manually fixed (by removing code) some new problems appear, which leads me to think that the avr assembler is not used.

If this is specific to my setup I will be happy to attach any details.

jantje, 2017-02-16:
Oeps I forgot to update the market place.
Lots of changes were done in V4 to add support for lots of boards.

jantje, 2017-02-16:
I updated the marketplace

ligius-, 2017-02-17:
I removed the old plugin, updated to 4.0, reconfigured the plugin but the issues above still remain.
I've created a pastebin with the output from Arduino IDE vs. Eclipse: http://pastebin.com/QUdvFYP8

jantje, 2017-02-17:
Thanks for the pastebin info.
From that info you can see that the Arduino IDE does not try to compile usbdrvasm.asm.
I do not know what the current criteria are the Arduino IDE uses to not compile and use the usbdrvasm.S file directly.
Anyway you can tell Sloeber to use the S file instead of the .asm file in 2 ways.
in the project explorer go to you project/libraries/DigiSparkMouse
Right click the usbdrvasm.asm file and select
1) delete to fix this problem for this sloeber installation.
2)resource configuration->exclude from build->select all->ok to fix the problem for your current project

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2018-04-20:
HELP!!
Can someone having this issue find out what the compilation rules are by Arduino IDE?

jantje, 2018-06-19:
I'm just wondering..could it be Arduino IDE does not  compile .asm files?

jantje, 2020-06-30:
I think this issue is close

closed 2020-06-30 by jantje