Sloeber/arduino-eclipse-plugin #661: An option to configure AVRDUDE command line parameters

bkeevil, 2017-02-20:
I'm not getting any info about how much RAM my program is using.  On the Arduino.cc IDE you can have AVRDUDE produce verbose output to get such info.  It would be nice if this plugin could do that too.

jantje, 2017-02-20:
@bkeevil 
I'm not sure what you refer to. At the end of compilation there is the size and the alternative size calculation in Sloeber.
Arduino IDE does something extra to account for the bootloader.
Is it this you are referring to or something else?

bkeevil, 2017-02-20:
It says how many bytes of flash have been uploaded but I can't find any other data.  I'm not sure what the "alternative size calculation in Sloeber" is?  

I am running Linux Mint, maybe the display is different?  

What I would like to see is something like what is in the regular Arduino IDE, where it says something like:

> Sketch uses 8234 bytes (3%) of program storage space. Maximum is 253952 bytes.
> Global variables use 367 bytes (4%) of dynamic memory, leaving 7825 bytes for local variables. Maximum is 8192 bytes.

There is also a screen with this sort of info produced by AVRDUDE in verbose mode (I think).  How can I change the options being sent to AVRDUDE?  It looks like in at least one version of this program that was possible to do, but not in the version I just downloaded from Marketplace.

jantje, 2017-02-20:
As usual things are far more complex than the average user expects. In this particular case you are referring to the size recipe as defined in the platform.txt file
```
## Compute size
recipe.size.pattern="{compiler.path}{compiler.size.cmd}" -A "{build.path}/{build.project_name}.elf"
recipe.size.regex=^(?:\.text|\.data|\.bootloader)\s+([0-9]+).*
recipe.size.regex.data=^(?:\.data|\.bss|\.noinit)\s+([0-9]+).*
recipe.size.regex.eeprom=^(?:\.eeprom)\s+([0-9]+).*
```
As you see there are some regular expressions there that extract information out of the output which is input for a custom java calculation taking into account the bootloader properties to provide the information you are looking for.
Sloeber only provides the output of the avr-size command. This is so because you can upload without bootloader, may have a different bootloader, know how big the bootloader is .....

For avr boards there is an alternative size command available in project properties->arduino->compile options. This one looks more like the arduino output but still doesn't account for the bootloader.
You can change the platform.txt file if you know a better command.
Please share the command if you find it :-)

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jrobert-gh, 2017-05-03:
When I execute a Verify, the console output includes something like this:
```
section                     size      addr
.data                        362   8388864
.text                     28174                0
.bss                          981   8389226
```
, where the sum of .data and .bss is the initial amount of RAM allocated to the program.
I believe the compilation that (optionally) takes place before an upload displays it too, but it flashes by too quickly to see it.

Gord1, 2017-09-18:
Yes, it would be nice if Sloeber could identify Program and RAM space used.

For other newbees that come across this.
Flash Program Space = .data + .text
RAM = .data + .bss

For Example:

Arduino IDE
Sketch uses 2270 bytes (7%) of program storage space. Maximum is 30720 bytes.
Global variables use 294 bytes (14%) of dynamic memory, leaving 1754 bytes for local variables. Maximum is 2048 bytes.

Sloeber
section                     size      addr
.data                        128   8388864
.text                       2142         0
.bss                         166   8388992

Premik, 2018-03-06:
Enabling the aforementioned alternative size option in the settings produces % of used space:
```
Device: atmega88p

Program:     502 bytes (6.1% Full)
(.text + .data + .bootloader)

Data:          7 bytes (0.7% Full)
(.data + .bss + .noinit)
```
Which I believe is good enought. The option is little bit "hidden" or at least I couldn't find it first. But well...

Gurkengewuerz, 2020-10-14:
Any updated on this? 👀

jantje, 2020-10-14:
The update is that there has not been any progress.

jantje, 2021-09-03:
[awk_GNU_64_bit_V4_2_1.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/7102568/awk_GNU_64_bit_V4_2_1.zip)
I''m planning on using awk to parse the size command output.
Anyone has any experiance with awk install on linux and or mac?

closed 2021-09-10 by jantje