Sloeber/arduino-eclipse-plugin #662: can FreeRTOS be used with the ESP8266 in sloeber.

ghislainborremans, 2017-02-20:
Is there an example of FreeRTOS in sloeber?

jantje, 2017-02-20:
Not that I'm aware of

ghislainborremans, 2017-02-22:
FreeRTOS is in the libraries for the ESP8266 platform but it refers to AVR\io.h... . I suppose it is not meant to be there.

jantje, 2017-02-22:
I had a look here and freertos is a library you can install in windows->preferences->arduino->library manager.
Once installed there are 3 examples available.
I tried analogread_digitalread and that compiles fine with the uno if you add the library manually.
Adding the library manually is needed because the include directive states Arduino_FreeRTOS.h and the foldername is FreeRTOS.
So yes there are examples for free rtos

ghislainborremans, 2017-02-22:
I understand. i was looking to have it on ESP8266. 
I have now switched to  "taskscheduler".  ( https://github.com/arkhipenko/TaskScheduler)

jantje, 2017-02-22:
The question "can freertos libary work with ESP8255" is for https://github.com/greiman/FreeRTOS-Arduino to answer. Not for sloeber.

ghislainborremans, 2017-02-22:
ok

closed 2017-02-22 by ghislainborremans