Sloeber/arduino-eclipse-plugin #663: new sketch based on example==>..\.ino.cpp:7:21: fatal error: Arduino.h: No such file or directory  #include "Arduino.h"

ghislainborremans, 2017-02-22:
When creating a new sketch based on the examples it is very important to rename the project and the ino file. 
If you keep the names of the sample you well get the error below.

 .\.ino.cpp:7:21: fatal error: Arduino.h: No such file or directory
 #include "Arduino.h"

best regards

jantje, 2017-02-22:
I create new sketches all the time based on examples and I never rename the .ino files.
Her is how I do it. https://youtu.be/MGAyIOC24lU?t=197
Can you elaborate on what you are doing differently?

ghislainborremans, 2017-02-22:
Some more testing: 

 if you rename the project, then you get the error 

..ino.cpp:7:21: fatal error: Arduino.h: No such file or directory
#include "Arduino.h"

I changed the project name back to the name of the example file and that generated the problem..

jantje, 2017-02-22:
That is something I can test and I think I even know the root cause before I test.
I did some optimisation (the makefiles were constantly regenerated) and I think I optimized to much.

victorpv, 2017-03-04:
I have the same problem when I copy a project into another.
If I got to Project->Properties->Arduino, and then click on "apply" at the bottom of "Arduino Board Selection" it resolves the problem.

jantje, 2017-03-04:
Seems like the same optimisation that is causing this.

jantje, 2017-03-06:
I confirm that renaming and copying causes the build to fail.
Workaround project properties->Arduino->apply->ok.
If needed also Arduino->reattach libraries and/or delete the release folder

jantje, 2017-03-07:
I have been investigating this and the sub parts of the recipe are being expanded before being stored.
This is so because some commands that need modification are in other environment variables making for the need of expansion.
This is not an issue except for the "linking" because the output is the FQN which includes the (old)project name after a rename/copy.
There are 2 ways around this.
1) do the modification in a non-expansioned way (kind of like recursive all used fields)
2) Do the reassigning when the project is created/copied
1 is to error prone as to my likings
2 needs an extra listener or a field to see the "project name during assignment"
I'm going for 2 with the extra field.

jantje, 2017-03-13:
Anyone wanting to test the fix?

ghislainborremans, 2017-03-15:
First test looks ok. 
I will test it further this week.

jantje, 2017-03-15:
txs :-)

ghislainborremans, 2017-03-20:
I ran this now a few days and I did no longer encounter this problem.

jantje, 2017-03-20:
txs :-)

closed 2017-05-25 by jantje