Sloeber/arduino-eclipse-plugin #664: Unterminated string at line 2999 column 80

skcks, 2017-02-22:
!MESSAGE An internal error occurred during: "Finishing the installation ..".
!STACK 0
com.google.gson.JsonSyntaxException: com.google.gson.stream.MalformedJsonException: Unterminated string at line 2999 column 80
	at com.google.gson.Gson.fromJson(Gson.java:818)
	at com.google.gson.Gson.fromJson(Gson.java:741)
	at io.sloeber.core.managers.Manager.loadLibraryIndex(Manager.java:278)
	at io.sloeber.core.managers.Manager.loadIndices(Manager.java:191)
	at io.sloeber.core.managers.Manager.startup_Pluging(Manager.java:92)
	at io.sloeber.core.Activator$2.run(Activator.java:199)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
Caused by: com.google.gson.stream.MalformedJsonException: Unterminated string at line 2999 column 80
	at com.google.gson.stream.JsonReader.syntaxError(JsonReader.java:1505)
	at com.google.gson.stream.JsonReader.nextQuotedValue(JsonReader.java:1006)
	at com.google.gson.stream.JsonReader.nextString(JsonReader.java:811)
	at com.google.gson.internal.bind.TypeAdapters$13.read(TypeAdapters.java:358)
	at com.google.gson.internal.bind.TypeAdapters$13.read(TypeAdapters.java:346)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$1.read(ReflectiveTypeAdapterFactory.java:93)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:172)
	at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.read(TypeAdapterRuntimeTypeWrapper.java:40)
	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.read(CollectionTypeAdapterFactory.java:81)
	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.read(CollectionTypeAdapterFactory.java:60)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$1.read(ReflectiveTypeAdapterFactory.java:93)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:172)
	at com.google.gson.Gson.fromJson(Gson.java:803)
	... 6 more

[image]
[image]

jantje, 2017-02-22:
This error is caused by a download of a json file that can not be parsed.
I have seen this happen in V3 when the json file was referenced but never on a line number as high as 2999.
As far as I remembered I added info on which file caused the problem in V4. As I don't see this info: Are you on V4?

skcks, 2017-02-27:
yes, it's v4.

jantje, 2017-02-27:
It looks to me as if the [eclipse install]/arduinoPlugin/library_index.json is corrupt on your system.
Mine is 2.234 kB and works fine.
Can you delete the file and restart sloeber?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

skcks, 2017-02-28:
it work, thanks for your help.  I download files(library_index.json, package_index.json) to the dir: /Applications/Eclipse.app/Contents/Eclipse/arduinoPlugin/, and extract avr-gcc-4.9.2-atmel3.5.3-arduino2-i386-apple-darwin11.tar.bz2 to the dir :     /Applications/Eclipse.app/Contents/Eclipse/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2

jantje, 2017-02-28:
Thanks for the feedback. Glad it helped.
Please consider becoming a patron as i really need the financial support. https://www.patreon.com/jantje

I still need to make this a nicer message so I reopen the issue

jantje, 2017-03-13:
Can you test whether you get a nicer message now?
Easiest is rename the file start sloeber. evaluate the test. Stop Sloeber put the file back.

jantje, 2017-04-24:
@skcks 
Can you test whether you get a nicer message now?
Easiest is rename the file start sloeber. evaluate the test. Stop Sloeber put the file back.

closed 2017-05-25 by jantje