Sloeber/arduino-eclipse-plugin #665: Eclipse J-Link install fails.

larnold7, 2017-02-22:
How do I debug DUE? worked last week. with 3.X. works on windows with 4.0.0.201701171403

When trying to install J-Link debugger from GNU arm eclipse plugins, I receive an error:

Only one debug restart button can be installed at once.

I don't know how to configure things with just OPENOCD. Can you give me a hint? or is it possible to remove OPENOCD and reinstall from GNU arm eclipse?
Thanks. nice product once you get over the learning curve.

jantje, 2017-02-23:
You probably have this issue because part of GNU arm eclipse plugins now comes with the sloeber product.
I see following possible root causes:
1)J-link debugger is already installed
2) you use another update site (and thus version) than sloeber does (caused by problems with source forge)

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

larnold7, 2017-02-23:
Ahh. J-link is not installed ( it is not available for selection) 
Openocd is, but it shows up in the add new software window anyway.

What update site do you use?

jantje, 2017-02-23:
I'm using
http://sloeber.io/gnuarmeclipse/stable/
It should be in the available update sites.
It might be a version behind. Which may be your problem

larnold7, 2017-02-23:
I tried http://sloeber.io/gnuarmeclipse/stable/    Same result on both Centos 7 and windows 7

error messages below. I'll try reverting to 3.X for now... got a big project

[j-link install error.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/796773/j-link.install.error.txt)

jantje, 2017-03-06:
I'm running the nightly so I should have used the nightly.  http://sloeber.io/gnuarmeclipse/stable/ allowed me to install.

closed 2017-05-25 by jantje