Sloeber/arduino-eclipse-plugin #667: c or ino files are not linked when the source files are in project subfolders

quicktom, 2017-02-26:
win32 sloeber 4.0

jantje, 2017-02-26:
As to my test ino files are not processed (both compiled and linked) but .c files are.
Currently .ino files are only expected in the root. Is this a problem?

```
'Building file: ..\test\test.c'
'Starting C compile'
"D:\embedded\V4\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"D:\embedded\V4\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.17\cores\arduino" -I"D:\embedded\V4\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.17\variants\standard" -I"D:\embedded\V4\sloeber\arduinoPlugin\libraries\FreeRTOS\9.0.0-1" -I"D:\embedded\V4\sloeber\arduinoPlugin\libraries\FreeRTOS\9.0.0-1\src" -MMD -MP -MF"test\test.c.d" -MT"test\test.c.o" -D__IN_ECLIPSE__=1 "..\test\test.c"  -o  "test\test.c.o"
'Finished building: ..\test\test.c'
' '
'Starting combiner'
"D:\embedded\V4\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o "D:/workspaces/embedded/freertos/Release/freertos.elf"    .\.ino.cpp.o  .\test\test.c.o  .\libraries\FreeRTOS\src\croutine.c.o .\libraries\FreeRTOS\src\event_groups.c.o .\libraries\FreeRTOS\src\heap_3.c.o .\libraries\FreeRTOS\src\list.c.o .\libraries\FreeRTOS\src\port.c.o .\libraries\FreeRTOS\src\queue.c.o .\libraries\FreeRTOS\src\tasks.c.o .\libraries\FreeRTOS\src\timers.c.o .\libraries\FreeRTOS\src\variantHooks.cpp.o   D:\workspaces\embedded\freertos\Release\arduino.ar    D:/workspaces/embedded/freertos/Release/arduino.ar  "-LD:/workspaces/embedded/freertos/Release" -lm
'Finished building: freertos.elf'
```

quicktom, 2017-02-27:
Hi jantje,

sorry, for me it is not working. I tried the same ab ubuntu, same result.
But if I rename c-files to cpp-files it working nicely.

jantje, 2017-02-27:
That is very strange behaviour.
Can you post a image of you project with the folder and the files visible in the project explorer?

I would try to delete the configuration build folder. This folder is normally called release. And build again.
Then create a new project and see whether the same happens there.
If you can reproduce in a new project, I would check windows->preferences->C/C++->file types. Check all filters if they would match your file.
If you can not reproduce the problem in another project you have probably been playing with resource configurations. Check in the problem project properties->C/C++general->Path and symbols->source location click on your project in the source and build path and click edit filter. There should only be filters starting with libraries

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

quicktom, 2017-02-27:
This is what's happening:
[image]

Renaming TestStepper.c to TestStepper.cpp makes the difference: 
[image]

windows->preferences->C/C++->file types:
[image]

jantje, 2017-02-27:
From the images I see that the folder has the same name as the project.
When I created a setup like that Sloeber started to do strange.
After deleting the Release folder doing a compile gives
```
22:06:25 **** Incremental Build of configuration Release for project freertos ****
"D:\\embedded\\V4\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
make: Nothing to be done for 'all'.

22:06:25 Build Finished (took 598ms)
```
As soon as I renamed it to something else (I used test and haha) all works fine again.
Seems like a folder in the root of a project with the same name as the project causes issues.

jantje, 2017-03-06:
Can you confirm that the root cause is the foldername which is the same as the project name?

closed 2017-05-25 by jantje