Sloeber/arduino-eclipse-plugin #669: Non closeable "become a patreon" window, window is larger than monitors resolution.

SzDavidHUN, 2017-03-04:
Linux Mint 18.1 Cinnamon, 1440×900+1600×900 (native resolution: 1440×900+1920×1080) screen.
The patreon popup screen can not be closed. The X button on the corner of the window not working, can't resize, can't close from task bar (no button shows for it), the OK/CANCEL are only accessable  on 1920×1080 monitors. Can't interact with the main window. So force close eclipse without able to save my work. Bonus: I don't know how to find which PIDs belongs to this eclipse, so have to shutdown all java based apps, "killall java", restart all.

jantje, 2017-03-04:
That is bad. There should be a close button on the bottom.
If this happens again can you take a screenshot?
I disabled the normal close because originally you could only close the screen after some time.

SzDavidHUN, 2017-03-07:
Here is the screenshot with 1600×900 screen. The same as in the 1440×900.
The left monitor is cutted off from the screenshot, nothing interesting there.

[image]

In 1920×1080 works.

[image]

jantje, 2017-03-08:
Apparently the toolbar at the bottom is not taken into account.
Do you get this page directly? You should see
[image]
As it is no longer needed I'll reenable to top cross

SzDavidHUN, 2017-03-08:
I saw that screen but forgot to take screenshot. Only then I remembered to take screenshot when I clicked on the patreon button.

jantje, 2017-03-08:
Tomorrow the nightly should allow to close with the top right X
I hope to see you as a patreon one day :)

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2017-03-13:
This is fixed

closed 2017-05-25 by jantje