Sloeber/arduino-eclipse-plugin #673: compiler.warning_flags from platform.txt is always added in the compile command

keestux, 2017-03-06:
Hi,

This is about compiler warning options for Arduino SAMD. They come from the platform.txt.

    compiler.warning_flags=-w
    compiler.warning_flags.none=-w
    compiler.warning_flags.default=
    compiler.warning_flags.more=-Wall
    compiler.warning_flags.all=-Wall -Wextra

In Arduino IDE you can select one of four, in Sloeber we have just a checkbox. Sloeber selects either the `.none` or the `.more`, I think. The selected option is added at the end of the compile command. The problem is that the compiler flags from `platform.txt` already contain `compiler.warning_flags`:

    ... -mthumb -c -g -Os {compiler.warning_flags} -std=gnu11 ...

For this Sloeber will replace `{compiler.warning_flags}` with `-w` as given in `platform.txt`. The result is that the compiler suppresses all warnings (even if `-Wall` is present).

jantje, 2017-03-06:
The implementation of warnings in sloeber is from before the Arduino IDE started using this method.
As such the code is somewhat outdated.
But this is how sloeber works (or at least is supposed to).
If the warning flag is not on then just use the platform.txt command.
If it is on replace the -w command with -wall. At least in V4 there used to be other implementations.

keestux, 2017-03-06:
> The implementation of warnings in sloeber is from before the Arduino IDE started using this method.
As such the code is somewhat outdated.

OK, I see.

> But this is how sloeber works (or at least is supposed to).
If the warning flag is not on then just use the platform.txt command.
If it is on replace the -w command with -wall. At least in V4 there used to be other implementations.

Well, that is not what I see.
If the warning flag is off you get:

    ... -mthumb -c -g -Os -w -std=gnu11 ... -w

If the warning flag is on you get:

    ... -mthumb -c -g -Os -w -std=gnu11 ... -Wall

In both cases there is a `-w` which turns off all compiler warnings, even with `-Wall`.

jantje, 2017-03-06:
You are probably not running V4 but V3.X

keestux, 2017-03-06:
Nope. I'm at V4

keestux, 2017-03-06:
Sloeber, the Eclipse Arduino IDE	4.0.0.201612232137	io.sloeber.product	null

jantje, 2017-03-06:
Strange I just tested it as I thought the -wall was at the end :-s

jantje, 2017-03-06:
It may have been I did in the nightly as I recall having had issues in this area and I had some serious work with / versus \ with the new make.exe on windows
```

'Starting C++ compile'
"D:\embedded\V4\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_YUN -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8041 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Yun\""  -I"D:\embedded\V4\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.17\cores\arduino" -I"D:\embedded\V4\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.17\variants\yun" -I"D:\embedded\V4\sloeber\arduinoPlugin\libraries\FreeRTOS\9.0.0-1" -I"D:\embedded\V4\sloeber\arduinoPlugin\libraries\FreeRTOS\9.0.0-1\src" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\.ino.cpp"  -o  ".ino.cpp.o"
'Finished building: ..\.ino.cpp'
```

```
'Building file: ..\.ino.cpp'
'Starting C++ compile'
"D:\embedded\V4\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_YUN -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8041 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Yun\""  -I"D:\embedded\V4\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.17\cores\arduino" -I"D:\embedded\V4\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.17\variants\yun" -I"D:\embedded\V4\sloeber\arduinoPlugin\libraries\FreeRTOS\9.0.0-1" -I"D:\embedded\V4\sloeber\arduinoPlugin\libraries\FreeRTOS\9.0.0-1\src" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\.ino.cpp"  -o  ".ino.cpp.o"
'Finished building: ..\.ino.cpp'
```

keestux, 2017-03-06:
I ran an update of the plugin and now it is OK.

Maybe I was running a beta version. Now I have

    Sloeber	4.0.0.201701171403	io.sloeber.feature.feature.group	Jan Baeyens and Others

    Sloeber, the Eclipse Arduino IDE	4.0.0.201612232137	io.sloeber.product	null

Thanks

closed 2017-03-06 by keestux

jantje, 2017-03-06:
Thanks for the input

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje