Sloeber/arduino-eclipse-plugin #675: serial monitor font preferences

atanisoft, 2017-03-07:
It would be great if the preferences could be enhanced to expose font preferences for the serial console.  It would be useful to switch to a monospace font as example for some devices which output formatted text (space or tab indents etc)

jantje, 2017-03-07:
you can set them to anything you want

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

in window->preferences->general->Appearance->colors and fonts->Sloeber->Serial Monitor

closed 2017-03-07 by jantje

atanisoft, 2017-03-07:
Thanks, I did not think to look under the General section.  I was expecting the Sloeber prefs to be under the Arduino section..

zoutepopcorn, 2019-01-26:
I changed my ~/.arduino15/preferences.txt ant changed:
```
editor.font=Cantarell,plain,12
serial.font=Cantarell,plain,12
```

I dont know if serial.font did the trick, but before only my code font was changed.