Sloeber/arduino-eclipse-plugin #676: Recipe for target "Clean" failed

Mingybear, 2017-03-07:
I have a fairly complex ESP8266 project with a number of source files. This file compiled, etc., fine under an older version of the plug-in. 

I set the project aside for a few months and started working on it again. I checked Eclipse for software updates and my Eclipse blew up completely (I suspect this was because I had tried Microsoft Bash but that is another issue. 

I reinstalled the whole package (i.e. Sloeber V4) and went down a rabbit hole of issues which I believe are Eclipse related. Since I was moving to a new laptop I installed Sloeber V4 on the new laptop which has never has Bash installed. 

My new laptop has Eclipse CDT, Sloeber V4 standalone (i.e. two separate programs), and MingW plus, of course, a few other packages like KiCad, etc.. Eclipse CDT compiles a Windows application project I made with no complaints. I used to have bother Eclipse CDT and V3 of Sloeber (before it was called Sloeber) on the other PC.

However when I compile my Arduino/ESP8266 project (which worked fine under V3 prior to Eclipse blowing up) I get the errors 

make: [clean] Error 1 (ignored)	PingerMar4		 	C/C++ Problem
recipe for target 'clean' failed	makefile	/PingerMar4/Release	line 97	C/C++ Problem

All the time without exception.

The line it is complaining about is the bolded text below.  I appended the make file - it is line 97

# Other Targets
clean:
	**-$(RM) $(C_UPPER_DEPS)$(PDE_DEPS)$(C_DEPS)$(AR)$(CC_DEPS)$(AR_OBJ)$(C++_DEPS)$(LINK_OBJ)$(CXX_DEPS)$(ASM_DEPS)$(HEX)$(INO_DEPS)$(SIZEDUMMY)$(S_UPPER_DEPS)$(ELF)$(CPP_DEPS)**
	-@echo ' '

I am at my wits end trying to solve this issue. I have tried to reinstall . Prior posts about this problem suggested installing the most recent version but the version I installed was March 6 so there must be another issue.

Any suggestions would be welcome. Thanks

Makefile
################################################################################
# Automatically-generated file. Do not edit!
################################################################################

-include ../makefile.init

RM := del 

# All of the sources participating in the build are defined here
-include sources.mk
-include subdir.mk
-include libraries\ESP8266WiFi\src/subdir.mk
-include libraries\ESP8266WebServer\src/subdir.mk
-include core\core\umm_malloc/subdir.mk
-include core\core\spiffs/subdir.mk
-include core\core\libb64/subdir.mk
-include core\core/subdir.mk
-include objects.mk

ifneq ($(MAKECMDGOALS),clean)
ifneq ($(strip $(CC_DEPS)),)
-include $(CC_DEPS)
endif
ifneq ($(strip $(C++_DEPS)),)
-include $(C++_DEPS)
endif
ifneq ($(strip $(C_UPPER_DEPS)),)
-include $(C_UPPER_DEPS)
endif
ifneq ($(strip $(CXX_DEPS)),)
-include $(CXX_DEPS)
endif
ifneq ($(strip $(ASM_DEPS)),)
-include $(ASM_DEPS)
endif
ifneq ($(strip $(INO_DEPS)),)
-include $(INO_DEPS)
endif
ifneq ($(strip $(PDE_DEPS)),)
-include $(PDE_DEPS)
endif
ifneq ($(strip $(S_UPPER_DEPS)),)
-include $(S_UPPER_DEPS)
endif
ifneq ($(strip $(CPP_DEPS)),)
-include $(CPP_DEPS)
endif
ifneq ($(strip $(C_DEPS)),)
-include $(C_DEPS)
endif
endif

-include ../makefile.defs

# Add inputs and outputs from these tool invocations to the build variables 
AR += \
C:/Users/Brian_17/Documents/ArduinoWorkspace/PingerMar4/Release/arduino.ar \

ELF += \
PingerMar4.elf \

HEX += \
PingerMar4.hex \

# All Target
all: PingerMar4

# Tool invocations
PingerMar4: $(HEX)
	@echo 'Building target: $@'
	@echo 'Printing size:'
	"C:\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-size" -A "C:/Users/Brian_17/Documents/ArduinoWorkspace/PingerMar4/Release/PingerMar4.elf"
	@echo 'Finished building target: $@'
	@echo ' '

C:/Users/Brian_17/Documents/ArduinoWorkspace/PingerMar4/Release/arduino.ar: $(AR_OBJ)
	@echo 'Starting archiver'
	"C:\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru "C:\Users\Brian_17\Documents\ArduinoWorkspace\PingerMar4/Release/arduino.ar"    C:/Users/Brian_17/Documents/ArduinoWorkspace/PingerMar4/Release/arduino.ar $(AR_OBJ)
	@echo 'Finished building: $@'
	@echo ' '

PingerMar4.elf: $(LINK_OBJ) $(AR)
	@echo 'Starting combiner'
	"C:\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -Wall -Wextra -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-LC:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lib" "-LC:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/ld" "-Teagle.flash.4m1m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy -o "C:\Users\Brian_17\Documents\ArduinoWorkspace\PingerMar4/Release/PingerMar4.elf" -Wl,--start-group   $(LINK_OBJ) $(AR)  "C:\Users\Brian_17\Documents\ArduinoWorkspace\PingerMar4/Release/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group "-LC:\Users\Brian_17\Documents\ArduinoWorkspace\PingerMar4/Release" C:/Users/Brian_17/Documents/ArduinoWorkspace/PingerMar4/Release/arduino.ar
	@echo 'Finished building: $@'
	@echo ' '

PingerMar4.hex: $(ELF)
	@echo 'Create eeprom image'
	"C:\sloeber\/arduinoPlugin/packages/esp8266/tools/esptool/0.4.9/esptool.exe" -eo "C:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/bootloaders/eboot/eboot.elf" -bo "C:/Users/Brian_17/Documents/ArduinoWorkspace/PingerMar4/Release/PingerMar4.bin" -bm dio -bf 40 -bz 4M -bs .text -bp 4096 -ec -eo "C:/Users/Brian_17/Documents/ArduinoWorkspace/PingerMar4/Release/PingerMar4.elf" -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
	@echo 'Finished building: $@'
	@echo ' '

# Other Targets
clean:
	-$(RM) $(C_UPPER_DEPS)$(PDE_DEPS)$(C_DEPS)$(AR)$(CC_DEPS)$(AR_OBJ)$(C++_DEPS)$(LINK_OBJ)$(CXX_DEPS)$(ASM_DEPS)$(HEX)$(INO_DEPS)$(SIZEDUMMY)$(S_UPPER_DEPS)$(ELF)$(CPP_DEPS)
	-@echo ' '

.PHONY: all clean dependents
.SECONDARY:

-include ../makefile.targets

jantje, 2017-03-07:
The clean not working is annoying but not a real problem. Just delete the Release folder to work around.

Though you say you are running Sloeber product V4 from march 6 release it doesn't look like it because you have 
`-include libraries\ESP8266WiFi\src/subdir.mk`
it should be
`-include libraries\ESP8266WiFi\src\subdir.mk`
Except for some executable names al should be \ on windows and / in linux/mac
Are you trying to run the linux version on windows?

There really is no need to install CDT next to sloeber. Sloeber contains standard CDT so having 2 eclipses is kind of overkill.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

Mingybear, 2017-03-07:
Jantje
Thank you so much for your prompt reply. I downloaded that version by clicking the button "Download 64 bits bundle" from http://eclipse.baeyens.it/stable.php?OS=Windows

I deleted the release directory and now get some weird compile time errors I never saw before. 

I will install the more recent build and see if it makes a difference.

Thanks again

Mingybear, 2017-03-07:
Update

I used the "latest build update" URL and updated the Sloeber package through Eclipse. 

I can now build my project so this particular issue is closed.

I may have another issue however: I wanted to incorporate a "send gmail" function into my code and that causes another issue which I will investigate further.

So it seems that updating to the most recent nightly build (but not the "Stable Build" corrects that issue.

I'll post again if I can figure out if the other problem is software related or Sloeber related

Thanks again