Sloeber/arduino-eclipse-plugin #678: Sloeber v4: impossible to build ESPEasy branch mega

uzi18, 2017-03-08:
repo: https://github.com/letscontrolit/ESPEasy

pts. to reproduce:
Add new arduino project
Choose Nodemcu and rest of build options
import src sketch dir to project
Build/Verify

errors: no reference to setup/loop

In create-new-arduino-project wizard it is impossible to choose src dir as main source for new project.

jantje, 2017-03-08:
Sloeber wants you ino file to be in the root of the project.

uzi18, 2017-03-08:
So how to import existing arduino project into root folder?

jantje, 2017-03-08:
Your problem is not clear to me.
For instance

> import src sketch dir to project

Is this part of the create new sketch functionality? Or is this some action you are doing using default OS/eclipse/cdt actions?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

uzi18, 2017-03-08:
Yes,  I'm trying to add files into new project by importing it with wizard.
After all have one template ino file in root folder and src folder with
real arduino project

This does not build as compiler shows undefinied reference to setup and
loop functions.

Any workaround to use src subfolder files in project as root files?
I will try just drag and drop them into root project dir in eclipse.

https://www.patreon.com/jantje

uzi18, 2017-03-08:
No I have clean eclipse with Sloeber v4
I add new arduino project,  so i'm forced to use default ino template as
there is no option in wizard to choose needed dir.
So after all I use another sloeber wizard to import existing arduino
project.
End up with eclipse/sloeber project that don't link at all.

https://www.patreon.com/jantje

jantje, 2017-03-09:
It looks like the ino file is not compiled, but the info you provide is way to vague.
As you are not sponsoring the project you are kind of on your own.
From the faq 

> 
> How can I find out more of the problem at hand?
> 
> Thanks for asking.
> 
> It is always a good idea to find out more about the problem at hand.
> 
> There are actually 3 Eclipse views that can give you more feedback on the problem at hand.
> 
>     The console view
>     The problems view
>     The errors view
> 
> The console view gives you more information on the commands that are being run externally. That is the compiler and avr dude. The commands should be nearly 100% the same as in the Arduino IDE (the order is not always relevant). Differences I am aware of are
> 
>     -D__IN_ECLIPSE__=1: that specifies you are in Eclipse(I'm considering to remove this)
>     -Wall: stating that all warnings must be shown (the Arduino IDE does not show warnings)
>     -x c++: An option I added to force *.pde and *.ino to be treated as C++ files (the latest versions of Arduino do this also)
> 
> The problems view gives an overview of the code problems and is probably not what you are looking for when you are reading this.
> 
> The error view gives you an overview of errors and warnings that happened in Eclipse and or its plugins. To show the errors window: select Window->show view -> other->select general tab->error log. Look at the plugin column to see which plugin logged the warning. I can only tell you more about the errors and warnings generated by io.sloeber.* plugins.
> 
> A final and most time consuming way to find the problem is to download the code from github. You can then compile and debug the code.

closed 2017-03-09 by jantje

uzi18, 2017-03-10:
Don't understand why it is possible to import "Linked" source dir as files inside will not compile at all.

If You point me out where to search in plugin sources, I could try provide some fix/PR.