Sloeber/arduino-eclipse-plugin #679: Additional ESP libraries

atanisoft, 2017-03-09:
It would be great if these libraries could be added to the default available set:
https://github.com/me-no-dev/ESPAsyncWebServer
https://github.com/me-no-dev/ESPAsyncTCP
https://github.com/me-no-dev/ESPAsyncUDP

jantje, 2017-03-09:
These look like ESP specific libraries and as such should not be on the default list.
I'm considering adding libraries that are not found automatically which should fix this.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

atanisoft, 2017-03-10:
Yes they are ESP specific, is it possible to have a set of default libraries for ESP and a default for Arduino?

jantje, 2017-03-13:
>is it possible to have a set of default libraries for ESP and a default for Arduino?

Without serious sponsoring I'm not going there.
But importing libraries on usage is something I have on my wish list

jantje, 2017-09-17:
it is actually install as the library is imported in the project.

jantje, 2017-10-20:
Please look at issue #825 to set your cdt indexer setting for this feature to work fine.

closed 2017-10-20 by jantje