Sloeber/arduino-eclipse-plugin #680: New-project shouldn't add an ino template if importing an ArduinoIDE project

jrobert-gh, 2017-03-09:
If I have created a project with the ArduinoIDE, and import it to Sloeber with New Project (same name), the default(?) .ino template is added to the project folder. One isn't needed in this case.

jantje, 2017-03-09:
Can you specify the exact steps to reproduce?

jrobert-gh, 2017-03-09:
- Open ArduinoIDE and create a new sketch. Save it with a project name which doesn't exist within Sloeber. I used:
```
void setup() {
  // put your setup code here, to run once:
	Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
	Serial.println("Hello, world!\n");
}
```
, and called it hello.
 - In Sloeber, click the New Project button. Type the project name ("hello", in my case). Click Finish.
 - Expand "hello" with project explorer. See two .ino files: hello.ino and blink.ino; also blink.txt, but that won't break the compile. Blink.{ino,txt} were added during the import.

jantje, 2017-03-10:
If I understand correctly your workspace is the root of the arduino IDE sketches.
I'll try to reproduce on my system.

uzi18, 2017-03-10:
Same problem like #678.
If I import arduino project I expect to treat it as root project.

jantje, 2017-03-10:
@uzi18 
This is not the same problem.
1) you are not a patron of mine and @jrobert-gh is
2) #678 talks about a src folder and this one about a created ino file

jrobert-gh, 2017-03-10:
Yes, I use the same workspace for both IDEs and keep my projects compilable by either one. (I teach with the ArduinoIDE but when not in front of a class, and for real work, I use Sloeber).

jantje, 2017-03-10:
When a new project is created for a existing folder the files are created as selected in the wizard.
But no files are overwritten.
Normally the Arduino IDE created a [foldername].ino file. Sloeber does the same thing (when selecting default ino file). As such no file should be created.
When selecting "default cpp","template" or "sample sketch" this will not work. Shall I add "empty project" to the options?

jrobert-gh, 2017-03-11:
The previous (V3) behavior seemed ideal, if it's still possible: If an ArduinoIDE project exists in the workspace with the name given to Sloeber for a New Project, create nothing (except Sloeber's dot files, of course); just inherit the project as is.
Importing a project wouldn't have to require any more than the project name and 'Finish'. 'Continue' would still give the user access to board and port selection and the other options available for truly new projects. In the minimum case, the imported project's folder contents would appear unchanged; only that it would now show up in Project Explorer.
Is that workable?

jantje, 2017-03-11:
As far as I'm aware nothing changed between V3 and V4 in this regards. I checked the history and I have seen nothing that makes the "new sketch wizard with existing folder" behave different.
Are you sure that when you do this the code descriptor page states default ino?

jrobert-gh, 2017-03-11:
> Are you sure that when you do this the code descriptor page states default ino?
I'm not sure what that means - The Sketch Template Folder selection, 2d or 3d dialog in the New Project wizard? If so, I leave it selected to the template I normally use to create a new project in Sloeber. Just now I tested with it selected to BareMinimum. I created a project, "thisisatest" with the ArduinoIDE with a trivial sketch in it. After import, my thisisatest.ino was still there, but BareMinimum.ino and BareMinimum.txt had been added.

jantje, 2017-03-11:
It should be "default ino file" in the "provide sketch template folder" (what a bad name :-s)

jrobert-gh, 2017-03-12:
Yes - that fixed it. Thank you.
 Would it make sense for Sloeber to assume "default ino file" when importing (New Project, and project folder already exists) without breaking anything else? Is there a use-case for altering a project (even by the addition of files) during import?

On a side note, how does "provide sketch template folder" / "Custom Template" recognize a custom template? I can navigate to mine OK, but the files are grayed-out. I've tried naming then .ino and .cpp but neither is recognized. They worked fine, named ....ino, in Sloebers internal "examples" library but that isn't really the right place for them.

jantje, 2017-03-12:
Cool that the extra file is fixed :-)

>Would it make sense for Sloeber to assume "default ino file" when importing (New Project, and project folder already exists) without breaking anything else?

I do not call your use case importing. Though the standard eclipse import ->project (with existing folder name) is an import what Sloeber does when using the new sketch wizard is creating a new Sloeber project and adds the existing code to the project. It is similar but different and important to understand the behaviour of sloeber create sketch wizard.

The fact that "default ino file" serves your use case is because Sloeber assumes it is better not to overwrite files during "file additions to a project".
As "default ino files" copies a "sketch.ino" file to [project name].ino and [project name].ino already exists (because arduino IDE enforces this) no additional file is added to the project.
With other words "sloeber is not importing a project in your use case, Sloeber is : creating a project and does not overwrite files during project creation". 
Mathematically speaking: A the file level Sloeber creates the union of a Arduino IDE sketch (A) and a sloeber project(S). In default ino file S=A∩S and therefore no files are added.
If you get this you can read the next section ;-)

When using custom template all files in the template folder will be copied to your project. The file called sketch.ino will be copied as [project name].ino.
So if your template folder contains 1 file named sketch.ino  S=A∩S will be true again. 
In other words "using the create new sketch wizard with a template that only contains sketch.ino will serve your use case"

To answer your question "Would it make sense for Sloeber to assume "default ino file" when importing (New Project, and project folder already exists) ". 
No it would not make sense, it should be "empty project" which would be feasible but it would have side effects. These side effects would mean lots of questions or lots of code changes to work around these side effects and I wonder why? 
For instance: now, when you create your second sketch all the old settings are saved. When you create one on a existing folder: would the "saved setting" have to be "empty project" or the "value saved before creating the project"?

My question to you is: Is it so bad to have a extra file at import? You only have this problem once for each Arduino IDE created sketch. You do not even have it when you create your sketches in Sloeber.

> Is there a use-case for altering a project (even by the addition of files) during import?

I see no use case. Eclipse supports so many ways to add files to a project I don't see a need for this. But, as explained above this is not a import...

>On a side note, how does "provide sketch template folder" / "Custom Template" recognize a custom template?

There have been changes in this area. Are you sure you are on the latest nightly? I'm actually wondering whether it is useful to do a check at all.

jrobert-gh, 2017-03-12:
No, I'm on the 4.0 release.

>... whether it is useful to do a check at all.
Probably not, since 1) the initial file is only meant as a starting point to begin my project from; and if I screwed it up it's my own fault because 2) I created it in the first place and presumably knew what I was doing. :)

jantje, 2017-03-19:
I think we can close this issue.

jrobert-gh, 2017-03-19:
Yes, selecting Default ino does  open a project  without adding filles.

jantje, 2017-03-19:
txs

closed 2017-03-19 by jantje