Sloeber/arduino-eclipse-plugin #681: Build-console doesn't report all compiler errors in multi-file project

jrobert-gh, 2017-03-09:
I'm developing a set of libraries and a test-program to exercise them. When I compile, the console only lists the errors in project files. Errors in the libraries don't appear there. I have to visit each library file in the editor and look for red flags on the right margin to find those. This happens even if I've cleaned the project to force a full build.

jantje, 2017-03-09:
https://www.patreon.com/posts/there-are-bugs-5507246

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jrobert-gh, 2017-03-09:
I am a patron, Jantje: 'JRobert' there, 'jrobert-gh' here.

jantje, 2017-03-10:
:-) thanks for being a patron. I advice to mention that as there is no link between github an patreon and my brain is not build for this kind of tasks.
Did the link answer your question?

jrobert-gh, 2017-03-10:
I understand the difference between errors found by the indexer/Codan, and those found by the compiler but this looks like it should be one of the latter. 

Here's a for-instance: In my  mis-capitalized a call to a class function (from outside the class). It was correctly flagged in the margin by Sloeber, but the console log never mentioned it. The call isn't resolvable as I had mis-spelled it; No function, and certainly no member-function exists with that spelling.

Similarly, I declared another member function in library file foo.h, then defined it in foo.cpp but with a different return type. Again, it's something a compiler couldn't fail to notice. The two files together are one compilation unit so this error would be visible during semantic analysis; it wouldn't take until bind-time to find this one.

jantje, 2017-03-10:
That is not expected behaviour and also not what I see here.
As you can see I use libraries very intensively and do development on my libraries at the same time as I develop projects and I have seen "strange behaviour" when using libraries between projects on the same workspace because you have opened a library in one project and you think you are in another in combination with project level defines but it was always me misunderstanding what was going on.
Here is a screenshot of a project of mine where you see the libraries (most of them developed by me) where I changed a void to int and the compiler complains about it.
[image]

jrobert-gh, 2017-03-10:
I also had an "'ObjectFoo' was not declared in this scope" error in one of my files. That was the only compiler message on the console until commented-out the offending member call (for now). Once I did that, the next compile emitted the prototype error I mentioned earlier.

... Hmm, ArduinoIDE just did the same thing; the 'not declared in scope' error was the only error until I fixed it; then the protoype errors would appear. So this is maybe a gcc issue? Anyway, it doesn't seem to depend on Sloeber or its console. Sorry for the false alarm.

jantje, 2017-03-10:
I can't say from here whether it is gcc/Sloeber/???
I can say (from experience) that there are strange behaviours everywhere.
If you can reliable reproduce the issue, and you can share the code, it sure is interesting to investigate because these strange behaviours are hard to find and most developers want to fix them.
I guess you know this but first thing in C/C++ world when the build does strange is: build->clean; if that doesn't fix it delete the build folder (normally release) and if then it still happens a closer look is what is needed.
In your case (sloeber and Arduino IDE are sharing the same folder) it may be that a file was on the system avoiding a rebuild of a source file in both IDE's (though unlikely as they both build on a different place). 

Note that in sloeber you also can force a rebuild of all files in make by adding the -B option to the make command (project properties->C/C++ Build->Build command)
The make -d option will tell you why the file was not build but will also generate a lot of output.

jantje, 2017-03-13:
Any progress on this issue?

jrobert-gh, 2017-03-14:
No, there's too much code involved for me to divert right now. The errors no announced on the console are still caught and show up in the editor windows, and will eventually appear on the console when the one(s) that over-ride them get fixed, so it's not show-stopper.

jantje, 2017-03-14:
ok

jantje, 2017-04-24:
Rereading this I was wondering....
Are you aware the default compilation option is "stop of first error" which means the compilation will stop after the first file that contains an error. In other words when 2 files contain an error only the errors in 1 file will be reported.
If you want this behavior changed go to project properties->C/C++ properties in the behavior tab

jrobert-gh, 2017-04-24:
Thanks, Jantje. That may well be the issue.
I remembered having changed that setting to 'don't stop on first error' but now I see that it is a per-project setting. I'll look at that when the behavior happens again. 

Is there someplace to choose the initial project-settings for new projects?

jantje, 2017-04-24:
>Is there someplace to choose the initial project-settings for new projects?

The windows->preference settings are "workspace/eclipse install" level. But I didn't find this setting there.

jantje, 2017-05-25:
Can we close this issue?

jrobert-gh, 2017-05-26:
Yes, thanks.

closed 2017-05-26 by jantje