Sloeber/arduino-eclipse-plugin #682: V3- Make tool path error when Importing project with Git

thesoulkiller, 2017-03-16:
Hello team,
Appreciated what you've developed, good work !
There is an issue about importing project from Git. The android project is created in Windows, and pushed to a git server. Then an import from Linux host caused such issue. 

"A_Windows_PATH/make" all 
Cannot run program "A_Windows_PATH/make": Unknown reason

Error: Program "A_Windows_PATH/make" not found in PATH
....

Then I tracked the problem is actually caused by a path variable:
JANTJE.MAKE_LOCATION - "A_Windows_PATH"

Then I made JANTJE.MAKE_LOCATION - Undefined, and it built successfully.
Cheers,
-Baris

jantje, 2017-03-16:
This is nor really the same issue as this https://github.com/Sloeber/arduino-eclipse-plugin/issues/34 but it is the same root cause and the workaround is the same.
From michael:

> The issue is not fixed in itself but we've found a viable workaround to it:
> .project is in .gitignore
> a file .project.copy needs to be copied manually to .project the first time
> selecting the project's properties and re-selecting the board/com port, results in the absolute paths being replaced with the correct ones.
> 
> So it's not perfect yet but we can live with it so well that I foregot about it.

jantje, 2017-03-19:
Triggering a environment update when the "project changes from os" is not going to solve this issue as this will create a new .(c)project file each time. but it will enable to check in the .(c)project file in a smart way.
As the original issue is unsolvable I'll reassign this issue to detecting os change and acting upon it.

closed 2017-05-25 by jantje