Sloeber/arduino-eclipse-plugin #683: compilation of Redbear duo (version 0.3.1) fails because of space in the command

jantje, 2017-03-17:
```
02:35:20 **** Build of configuration Release for project 191_RedBear_Duo_native ****
"D:\\embedded\\eclipse\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\.ino.cpp'
'Starting C++ compile'
" D:\embedded\eclipse\/arduinoPlugin/packages/RedBear/tools/arm-none-eabi-gcc-redbear/4.9-2015-q3/bin/ arm-none-eabi-gcc" -DF_CPU=120000000 -DARDUINO=10609 -DDFU_BUILD_ENABLE -DINCLUDE_PLATFORM=1 -DLOG_INCLUDE_SOURCE_INFO=1 "-DLOG_MODULE_CATEGORY=\"app\"" -DMODULAR_FIRMWARE=1 -DMODULE_DEPENDENCY=4,2,9 -DMODULE_FUNCTION=5 -DMODULE_INDEX=1 -DMODULE_VERSION=9 -DPARTICLE=1 -DPARTICLE_NO_ARDUINO_COMPATIBILITY=0 -DPARTICLE_USER_MODULE -DPLATFORM_ID=88 -DPLATFORM_NAME=duo -DPLATFORM_THREADING=1 -DPRODUCT_FIRMWARE_VERSION=65535 -DPRODUCT_ID=88 -DRELEASE_BUILD -DRESET_AVRDUDE_FLASHER_SERIAL_SPEED=1200 -DSPARK=1 -DSPARK_PLATFORM -DSPARK_PLATFORM_NET=BCM43438A1 -DSTART_AVRDUDE_FLASHER_SERIAL_SPEED=19200 -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -DSTM32F2XX -DSTM32_DEVICE -DSYSTEM_VERSION_STRING=0.3.1 -DUSBD_PID_CDC=0xC058 -DUSBD_PID_DFU=0xD058 -DUSBD_VID_SPARK=0x2B04 -DUSER_FIRMWARE_IMAGE_LOCATION=0x80C0000 -DUSER_FIRMWARE_IMAGE_SIZE=0x40000 -DUSE_CS=A2 -DUSE_SPI=SPI -DUSE_STDPERIPH_DRIVER -DUSE_THREADING=0 -D_GNU_SOURCE -D_WINSOCK_H  "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/communication/src" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/dynalib/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/shared" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/duo" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/duo/libraries/btstack/port" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/duo/libraries/btstack/src" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/duo/libraries/btstack/src/ble" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/duo/libraries/btstack/src/classic" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/stm32" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/stm32f2xx" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/CMSIS/Device/ST/Include" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/CMSIS/Include" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/STM32_StdPeriph_Driver/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/STM32_USB_Device_Driver/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/STM32_USB_Host_Driver/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/STM32_USB_OTG_Driver/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/shared/STM32/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/shared/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/services/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/system/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/wiring/inc" "-ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/user/inc" -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -MD -MP -MF "D:/workspaces/junit-workspace/191_RedBear_Duo_native/Release/191_RedBear_Duo_native.o.d" -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc -fno-exceptions -fno-rtti -fcheck-new -std=gnu++11 -c -o  -I"D:\embedded\eclipse\arduinoPlugin\packages\RedBear\hardware\STM32F2\0.3.1\cores\RedBear_Duo" -I"D:\embedded\eclipse\arduinoPlugin\packages\RedBear\hardware\STM32F2\0.3.1\variants\RedBear_Duo" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\.ino.cpp"   ".ino.cpp.o"
subdir.mk:24: recipe for target '.ino.cpp.o' failed
process_begin: CreateProcess(NULL, " D:\embedded\eclipse\/arduinoPlugin/packages/RedBear/tools/arm-none-eabi-gcc-redbear/4.9-2015-q3/bin/ arm-none-eabi-gcc" -DF_CPU=120000000 -DARDUINO=10609 -DDFU_BUILD_ENABLE -DINCLUDE_PLATFORM=1 -DLOG_INCLUDE_SOURCE_INFO=1 -DLOG_MODULE_CATEGORY=\"app\" -DMODULAR_FIRMWARE=1 -DMODULE_DEPENDENCY=4,2,9 -DMODULE_FUNCTION=5 -DMODULE_INDEX=1 -DMODULE_VERSION=9 -DPARTICLE=1 -DPARTICLE_NO_ARDUINO_COMPATIBILITY=0 -DPARTICLE_USER_MODULE -DPLATFORM_ID=88 -DPLATFORM_NAME=duo -DPLATFORM_THREADING=1 -DPRODUCT_FIRMWARE_VERSION=65535 -DPRODUCT_ID=88 -DRELEASE_BUILD -DRESET_AVRDUDE_FLASHER_SERIAL_SPEED=1200 -DSPARK=1 -DSPARK_PLATFORM -DSPARK_PLATFORM_NET=BCM43438A1 -DSTART_AVRDUDE_FLASHER_SERIAL_SPEED=19200 -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -DSTM32F2XX -DSTM32_DEVICE -DSYSTEM_VERSION_STRING=0.3.1 -DUSBD_PID_CDC=0xC058 -DUSBD_PID_DFU=0xD058 -DUSBD_VID_SPARK=0x2B04 -DUSER_FIRMWARE_IMAGE_LOCATION=0x80C0000 -DUSER_FIRMWARE_IMAGE_SIZE=0x40000 -DUSE_CS=A2 -DUSE_SPI=SPI -DUSE_STDPERIPH_DRIVER -DUSE_THREADING=0 -D_GNU_SOURCE -D_WINSOCK_H -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/communication/src -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/dynalib/inc -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/inc -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/shared -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/duo -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/duo/libraries/btstack/port -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/duo/libraries/btstack/src -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/duo/libraries/btstack/src/ble -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/duo/libraries/btstack/src/classic -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/stm32 -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/hal/src/stm32f2xx -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/CMSIS/Device/ST/Include -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/CMSIS/Include -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/STM32_StdPeriph_Driver/inc -ID:\embedded\eclipse\/arduinoPlugin/packages/RedBear/hardware/STM32F2/0.3.1/cores/RedBear_Duo/firmware/platform/MCU/STM32F2xx/STM32_USB_Device_Driver/inc -ID:\embedded\eclipse\ [... cut ...] -ID:\embedded\eclipse\arduinoPlugin\packages\RedBear\hardware\STM32F2\0.3.1\variants\RedBear_Duo -MMD -MP -MF.ino.cpp.d -MT.ino.cpp.o -D__IN_ECLIPSE__=1 -x c++ ..\.ino.cpp .ino.cpp.o, ...) failed.
make (e=2): Het systeem kan het opgegeven bestand niet vinden.

make: *** [.ino.cpp.o] Error 2

02:35:20 Build Finished (took 184ms)
```

Note the space before "arm-none-eabi-gcc"
`" D:\embedded\eclipse\/arduinoPlugin/packages/RedBear/tools/arm-none-eabi-gcc-redbear/4.9-2015-q3/bin/ arm-none-eabi-gcc"`

jantje, 2017-03-17:
workaround
remove the space after the = in the platform.txt
```
tools.arm_none_eabi_gcc.path         = {runtime.tools.arm-none-eabi-gcc-redbear-4.9-2015-q3.path}/bin/
tools.arm_none_eabi_gcc.as           = arm-none-eabi-as
tools.arm_none_eabi_gcc.gcc          = arm-none-eabi-gcc
tools.arm_none_eabi_gcc.g++          = arm-none-eabi-g++
tools.arm_none_eabi_gcc.ar           = arm-none-eabi-ar
tools.arm_none_eabi_gcc.objcopy      = arm-none-eabi-objcopy
tools.arm_none_eabi_gcc.size         = arm-none-eabi-size
```

uzi18, 2017-03-17:
Works ok?

jantje, 2017-03-17:
I haven't tested it but you did what I expected to do :-)

uzi18, 2017-03-17:
Also didn't, but looks like it is root of the issue.
Just prepared Sloeber build env (PDE/JDT), but Run As .. still not work for me - need to investigate.

Also found some interesting issues, I'll try fill up new issues with minimal test case.

jantje, 2017-03-17:
There is also the junit tests you can run.
If you debug from the io.sloeber.product->META-INF->MANIFEST.MF it should work.
If it still doesn't work go to the debug configurations->your config->plugins->add required plugins

jantje, 2017-03-17:
Seems it doesn't work. Will need more trims :-(

jantje, 2017-03-18:
They also put the object file before the source file in platform.txt which makes the compile fail.
This  because sloeber assumes the source file is before the object file.

jantje, 2017-03-19:
so to work around the second issue you need to modify the platform.txt

uzi18, 2017-03-20:
is it possiblo to sort platform.txt before import?

jantje, 2017-03-20:
It is not a matter of sorting.
This gives a problem
`recipe.c.o.pattern                   = "{tools.arm_none_eabi_gcc.path}{tools.arm_none_eabi_gcc.gcc}" -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} {tools.arm_none_eabi_gcc.c.defines} {includes} {tools.arm_none_eabi_gcc.c.includes} {tools.arm_none_eabi_gcc.c.cflags} {tools.arm_none_eabi_gcc.c.conlyflags} -c -o "{object_file}" "{source_file}"`
this is fine
`recipe.c.o.pattern                   = "{tools.arm_none_eabi_gcc.path}{tools.arm_none_eabi_gcc.gcc}" -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} {tools.arm_none_eabi_gcc.c.defines} {includes} {tools.arm_none_eabi_gcc.c.includes} {tools.arm_none_eabi_gcc.c.cflags} {tools.arm_none_eabi_gcc.c.conlyflags} -c  "{source_file}" -o "{object_file}"`

uzi18, 2017-03-20:
where to investigate - any .java file?

jantje, 2017-03-20:
I really wouldn't know how to fix this

closed 2017-05-25 by jantje