Sloeber/arduino-eclipse-plugin #685: Including single library

hreintke, 2017-03-18:
When I configure the "Private Library path" with a path of a library like this 

[image]

I get the subdirectories in the list of libraries to include

[image]

This also shows up in the project library list : 

[image]

When both of the libraries have the same subfolder (in this case src) only one is available for the project.

jantje, 2017-03-18:
You need to add the parent of the libraries in other words the libraries folder.
In your case that would be C:\userdata.

hreintke, 2017-03-18:
I can to that but then I get a lot of "Arduino libraries" which are folders with completely other content.
What is the condition on which a subdirectory is considered as a library ?

jantje, 2017-03-19:
It is the arduino ide model ..../libraries/[your libraries]/examples
Sloeber wants to know .../libraries

hreintke, 2017-03-19:
I understand the ../libraries.
My question is on the subdirectories beneath that. 
My c:\userdata contains a lot of non-arduino subdirectories which have a complete other structure and no library.properties. But these do show up in the list of libraries when adding a library to the project.

jantje, 2017-03-19:
Sloeber doesn't care about library.properties. All subfolders are considered libraries

hreintke, 2017-03-20:
OK, clear

closed 2017-03-20 by hreintke