Sloeber/arduino-eclipse-plugin #688: Having troubles with FreeType2 library (on Windows)

nm3210, 2017-03-23:
I am trying to get a little I2C OLED screen working by using the following libraries:

- [Adafruit_SSD1306](https://github.com/adafruit/Adafruit_SSD1306) 
- [Adafruit_GFX](https://github.com/adafruit/Adafruit-GFX-Library)

It works/compiles/runs with the normal Arduino IDE just fine (1.8.0) using the libraries. But when I simply try to compile an empty sketch with those libraries as includes (so no code) within Sloeber, it barfs:

> 23:01:25 **** Incremental Build of configuration Release for project OledTesting ****
"C:\\Arduino\\sloeber 2017-01-17\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../OledTesting.cpp'
'Starting C++ compile'
"C:/Arduino/ArduinoIDE_1.8.0/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++" -c -Os --specs=nano.specs -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m0plus -fsingle-precision-constant -D__MKL26Z64__ -DTEENSYDUINO=134 -DARDUINO=10609 -DF_CPU=48000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH  -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\cores\teensy3" -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\libraries\Adafruit_GFX" -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\libraries\Adafruit_SSD1306" -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\libraries\SPI" -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\libraries\Wire" -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\libraries\Wire\utility" -I"C:\Arduino\Libraries\freetype-2.4.0\objs\.libs" -MMD -MP -MF"OledTesting.cpp.d" -MT"OledTesting.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../OledTesting.cpp"  -o  "OledTesting.cpp.o"
'Finished building: ../OledTesting.cpp'
' '
'Building file: C:/Arduino/ArduinoIDE_1.8.0/hardware/teensy/avr/libraries/Adafruit_GFX/fontconvert/fontconvert.c'
'Starting C compile'
"C:/Arduino/ArduinoIDE_1.8.0/hardware/teensy/../tools/arm/bin/arm-none-eabi-gcc" -c -Os --specs=nano.specs -g -Wall -ffunction-sections -fdata-sections -nostdlib -mthumb -mcpu=cortex-m0plus -fsingle-precision-constant -D__MKL26Z64__ -DTEENSYDUINO=134 -DARDUINO=10609 -DF_CPU=48000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH  -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\cores\teensy3" -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\libraries\Adafruit_GFX" -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\libraries\Adafruit_SSD1306" -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\libraries\SPI" -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\libraries\Wire" -I"C:\Arduino\ArduinoIDE_1.8.0\hardware\teensy\avr\libraries\Wire\utility" -MMD -MP -MF"libraries/Adafruit_GFX/fontconvert/fontconvert.c.d" -MT"libraries/Adafruit_GFX/fontconvert/fontconvert.c.o" -D__IN_ECLIPSE__=1 "C:/Arduino/ArduinoIDE_1.8.0/hardware/teensy/avr/libraries/Adafruit_GFX/fontconvert/fontconvert.c"  -o  "libraries/Adafruit_GFX/fontconvert/fontconvert.c.o"
C:/Arduino/ArduinoIDE_1.8.0/hardware/teensy/avr/libraries/Adafruit_GFX/fontconvert/fontconvert.c:23:22: **fatal error: ft2build.h: No such file or directory**
 #include <ft2build.h>
                      ^
compilation terminated.
libraries\Adafruit_GFX\fontconvert/subdir.mk:18: recipe for target 'libraries/Adafruit_GFX/fontconvert/fontconvert.c.o' failed
make: *** [libraries/Adafruit_GFX/fontconvert/fontconvert.c.o] Error 1
 &nbsp;
23:01:25 Build Finished (took 621ms)

I can understand that the Arduino IDE is probably going to use the included 'fontconvert_win.md', but i'm not exactly sure what I should do with Sloeber. I have tried adding a source folder ("C:\Arduino\Libraries\freetype-2.4.0\include") to my project and it then will compile for a little while but then complain again:

> 'Starting combiner'
"C:/Arduino/ArduinoIDE_1.8.0/hardware/teensy/../tools/arm/bin/arm-none-eabi-gcc" -Os --specs=nano.specs -Wl,--gc-sections,--relax,--defsym=__rtc_localtime=1490223539 "-TC:/Arduino/ArduinoIDE_1.8.0/hardware/teensy/avr/cores/teensy3/mkl26z64.ld" -mthumb -mcpu=cortex-m0plus -fsingle-precision-constant -o "C:\Users\Nate\Dropbox\Miscellaneous\GitRepo\OledTesting/Release/OledTesting.elf"    ./OledTesting.cpp.o  ./libraries/Wire/utility/twi.c.o  ./libraries/Wire/Wire.cpp.o  ./libraries/SPI/SPI.cpp.o  ./libraries/Adafruit_SSD1306/Adafruit_SSD1306.cpp.o  ./libraries/Adafruit_GFX/fontconvert/fontconvert.c.o  ./libraries/Adafruit_GFX/Adafruit_GFX.cpp.o ./libraries/Adafruit_GFX/glcdfont.c.o   C:/Users/Nate/Dropbox/Miscellaneous/GitRepo/OledTesting/Release/arduino.ar    C:/Users/Nate/Dropbox/Miscellaneous/GitRepo/OledTesting/Release/arduino.ar  "-LC:\Users\Nate\Dropbox\Miscellaneous\GitRepo\OledTesting/Release" -larm_cortexM0l_math -lm
./libraries/Adafruit_GFX/fontconvert/fontconvert.c.o: In function 'main':
C:/Arduino/ArduinoIDE_1.8.0/hardware/teensy/avr/libraries/Adafruit_GFX/fontconvert/fontconvert.c:115: **undefined reference to `FT_Init_FreeType'**

I presume this error is from it not being able to fine the correct binary? (forgive me if I'm incorrect) Does anyone know how to include the library correctly so this can work? Thanks!!

jantje, 2017-03-23:
The arduino IDE uses a "list of folders to use" with libraries.
CDT uses a "list of stuff to exclude from the build path"
These 2 are not really compatible so you will have to manually exclude folders from the build path by right clicking on the folder and selecting resource configurations->exclude from build
[image]

jantje, 2017-05-25:
duplicate of #642

closed 2017-05-25 by jantje