Sloeber/arduino-eclipse-plugin #689: Setting up plugin dev environment

hreintke, 2017-03-23:
Hi,
Try to setup an eclipse environment for building arduino-plugin.
Based on the description from the readme.

This is the resulting environment : 
Eclipse installed software : 

[image]

These are the project when including using git import

[image]

These are the error I get after import : 

[image]

Ant hints where I have made a mistake/missing parts ?

jantje, 2017-03-23:
Do you have the cdt build environment?

hreintke, 2017-03-23:
Thanks, I indeed missed the C++ tools.

Working as expected now.

closed 2017-03-23 by hreintke