Sloeber/arduino-eclipse-plugin #690: Git and problems moving between different pc environments

ghislainborremans, 2017-03-23:
I use sloeber and i use git. 
I work on different pc's and configurations. When i pull the version of another pc then the project is no longer considered a correct project and the build is no longer working. 
In the makefile some point to the  directory structure of the other pc. 
As the makefile is generated, it makes no sence to edit that. 
2 a) questions: what should ( or what should i not) push to git? 
b) can i define the variables for the makefile and what are the names for these variables.

jantje, 2017-03-23:
This is a duplicate of #34
This is the workaround

> The issue is not fixed in itself but we've found a viable workaround to it:
> .project is in .gitignore
> a file .project.copy needs to be copied manually to .project the first time
> selecting the project's properties and re-selecting the board/com port, results in the absolute paths being replaced with the correct ones.
>
>So it's not perfect yet but we can live with it so well that I foregot about it.

As far as I understand simply adding .project to the .gitignore will do the trick

Note that a fix for #682 will ease the pain even more (but .project still needs to be in .gitignore)

closed 2017-03-23 by ghislainborremans

jensriebold, 2017-05-24:
To ignore .project will lead to problems for other projects. 
.project should contain only relative paths or use eclipse environment variables. 

I have the problem that all my source code is stored in my SVN on a local network drive.
Mostly I develop with my linux pc and synchronize the work with my windows notebook when
I'm not at home.

Please fix the issue.... Thanks in advance!
Jens Riebold

jantje, 2017-05-24:
If you make a linked resource to all locations used; and are using the nightly; the .project only contains relative paths or start with environment variables.