Sloeber/arduino-eclipse-plugin #692: Trouble installing V4 on Linux

SKelem, 2017-03-24:
I have eclipse neon on a 64-bit Ubuntu Linux.
The installation instructions for V4 say:
    Follow the V3 instructions but use the V4 update site
So, I used the update site http://www.baeyens.it/eclipse/V4, but when that site is selected in Preferences/Available Software Sites, the message "This repository is currently not available." is displayed.

jantje, 2017-03-24:
The url is wrong
Can you point me to where you found this url?

jantje, 2017-03-29:
Can you point me to where you found this url?

jcollie, 2017-04-01:
@jantje the url @SKelem listed isn't found anywhere but the instructions under "Install V4 plugin" combined with the instructions on "Install V3 plugin" from this page:

http://eclipse.baeyens.it/how_to.shtml#/c1

Would lead one to believe that taking the URL http://www.baeyens.it/eclipse/V3 and replacing the 3 with a 4 gets you the URL for the V4 plugin install.

jantje, 2017-04-01:
@jcollie 
txs I'll fix that

jantje, 2017-04-04:
The web site has been updated

closed 2017-04-04 by jantje